EdgeTX/edgetx #795: 2.5 RC1 has a number of unexpected problems

MartinHugh, 2021-09-22:
I have installed this version and note the following anomalies when migrating a RadioMaster TX16S from OpenTX 2.3.14 to Edge TX 2.5 RC1 using the installation instructions given at https://github.com/EdgeTX/edgetx.github.io/wiki/EdgeTX-Installation-Guide

1.  The touch screen fails to work (it is a touch screen and I can verify it works under OpenTX, but does nothing under Edge TX)
2.  The RTN/cancel button has now replaced the MDL button. Pressing RTN does nothing
3.   Despite copying over my MODELS and RADIO folders from my OpenTx backup, I have no models to choose from once I start the radio.  Just a default Model1

Note that I tried an earlier release a month or two back and had none of these problems.

What am I missing ?

dannybgoode, 2021-09-22:
Are you sure you've installed the correct firmware as there are a few versions to choose from for different transmitters.

You need the one beginning tx16s 

I have 2.5.0 installed on my TX16S and it works absolutely fine.

MartinHugh, 2021-09-22:
@dannybgoode thank you for your reply. 
Yes you are right.  I had inadvertently selected the firmware starting "t16-" rather than "tx16s-"

It seems better now :-)

closed 2021-09-22 by MartinHugh