EdgeTX/edgetx #799: Audio confirmation of Custom Failsafe

MartinHugh, 2021-09-23:
Edge Tx 2.5 RC1
RadioMaster TX16S

I would be grateful if someone could verify whether this is an issue or, whether the problem is operator error.

When I set a custom failsafe in model settings :

Failsafe Mode/Set/
Scroll up one click to select "Channels=>Failsafe"
Long press OK

I expect audio feedback that the radio has accepted these setting to send to the RX

I am getting no audio (or visual feedback) that this has been done.

Thanks for reading

pfeerick, 2021-09-23:
I have only played with transmitter set failsafe settings a few times - both with OTX and ETX, and can't say I remember ever having any feedback of any sort... I've always checked afterwards that it was set.

MartinHugh, 2021-09-23:
@pfeerick, Just checked OpenTX 2.3.14.  On long pressing OK as described above, it gives **both** audio and haptic feedback.  
It makes sense to assume this should be carried forward to Edge, unless there is a good reason for omission ?

pfeerick, 2021-09-23:
No reason - probably just overlooked. Thanks for checking! :)

pfeerick, 2021-09-23:
Since this is so trivial a fix, I'll push it to 2.5 also once the checks complete.

closed 2021-09-23 by pfeerick

MartinHugh, 2021-09-23:
Many thanks @pfeerick

commit 580b9b12 (2021-09-23): Add missing audio/haptic when set custom failsafe

Fixes #799
commit f746d510 (2021-09-23): Add missing audio/haptic when set custom failsafe

Fixes #799