EdgeTX/edgetx #801: Potential issue with using module bay for updating receiver and module firmware

pfeerick, 2021-09-24:
**Describe the bug**
There have been a couple of reports now for 2.4.0 and 2.5.0-RC1, and I believe I have experienced this myself in testing a update related PR, that updating receivers (and in my case a XJT module) is currently not working. I have also not been able to update a ELRS module via the module bay. One user was able to update the receiver on the X7, but not the TX16S or X12s. 

Receivers:
- Radiomaster R168 RX 
- R-XSR

TX Module:
- Frsky XJT
- Happymodel ES24

Transmitters:
- RM TX16S
- X12s 

- https://www.rcgroups.com/forums/showpost.php?p=47869661&postcount=717
- https://discord.com/channels/839849772864503828/839849772864503832/890022344196767745
- https://discord.com/channels/839849772864503828/839849772864503832/890675015534665740

**To Reproduce**
Try to apply a firmware update via the module bay. Will most likely beep and 
```
Firmware update error 
Device not responding
```

**Expected behavior**
For the firmware update to complete!

wimalopaan, 2021-09-24:
The error on tx16s with latest main:
[image]

rotorman, 2021-09-24:
@wimalopaan can you please try to reload the image - the link in your post does not seem to work.

wimalopaan, 2021-09-25:
> @wimalopaan can you please try to reload the image - the link in your post does not seem to work.

Done

raphaelcoeffic, 2021-09-25:
> > @wimalopaan can you please try to reload the image - the link in your post does not seem to work.
> 
> Done

And also edit your post to specify which kind of firmware you were trying to upload, to which port, etc.

pfeerick, 2021-09-26:
XJT+TX16S => attempted update with Saleae monitoring SPORT => "Device not responding"
[XJT+TX16S-EdgeTX.zip](https://github.com/EdgeTX/edgetx/files/7231058/XJT%2BTX16S-EdgeTX.zip)

rotorman, 2021-09-26:
In both cases in Saleae Logic, the analyzer is thus 57600 baud inverted for Channel 0:
[image]
@pfeerick What does Channel 1 in the second trace show?

pfeerick, 2021-09-26:
PPM

PR #816 (2021-09-27): telemetry driver: clear TC flag before re-enabling TC interrupt
This prevents the USART TC IRQ handler to trigger right after the DMA request has completed, and thus turning the output OFF before the last byte has been transmitted.

Resolves #801

Tests:
- [x] TX16S (XJT + R9)
- [x] X7
- [x] TX12 (XJT)
- [x] X9D+2019 (XJT)
commit 07d0a663 (2021-09-27): telemetry driver: clear TC flag before re-enabling TC interrupt

This prevents the USART TC IRQ handler to trigger right after the DMA request has completed, and thus turning the output OFF before the last byte has been transmitted.

Resolves #801
commit e2a37f3f (2021-09-27): telemetry driver: clear TC flag before re-enabling TC interrupt

This prevents the USART TC IRQ handler to trigger right after the DMA request has completed, and thus turning the output OFF before the last byte has been transmitted.

Resolves #801
closed 2021-09-27 by raphaelcoeffic