EdgeTX/edgetx #802: Radioking TX18s: EdgeTx does not boot to UI if internal MPM is enabled.

Steini63, 2021-09-24:
Bug description:

Once the internal multiprotocol module is enabled for the active model, EdgeTx will not boot up to the UI. A black screen with the Enter symbol is displayed. If a switch or the throttle is not in the home position, the "Warning Screen" is the last display. Restart is only possible after removing the battery.

- In operation I can switch to the internal multiprotocol module, bind and use it normally. 
- Booting with a model **without** activated internal multiprotocol module works normally.
- Booting with a model with activated **external** multiprotocol module works normally.
- The latest firmware (that I own) that does **not** have the problem is 2.5.0 (8becbfcb) dated 2021-08-25.

General information:
 - EdgeTX version 2.5.0 RC1, 2.6.0 nightlies
 - Hardware: Radioking TX18s (flashed with Jumper T18 firmware)
 - internal Multiprotocol-Module on firmware 1.3.3.0 AETR  (1.3.2.86 AETR the same)

pfeerick, 2021-09-24:
Are you the same person who reported this today on RCGroups? If not, there is a similar report there

https://www.rcgroups.com/forums/showpost.php?p=47872683&postcount=729

pfeerick, 2021-09-24:
If you wouldn't mind, can you try this firmware build - it's from two weeks ago, just before the dynamic MPM protocol scanning was added... either way, it will start narrowing down when it horribly wrong for the RK TX18!

[t18-7fcc3a8 (3d0cedc).zip](https://github.com/EdgeTX/edgetx/files/7224951/t18-7fcc3a8.3d0cedc.zip)

Steini63, 2021-09-24:
No, I have not even noticed the post. In my case, "scanning for protocols" works without problems.

pfeerick, 2021-09-24:
Oh boy... ok, that sounds like it's something else entirely...

Steini63, 2021-09-24:
I will try the suggested build.

Steini63, 2021-09-24:
This one boots without issues.

raphaelcoeffic, 2021-09-24:
@Steini63 it would help greatly if you could upload your RADIO and MODELS directories and the name of the model that does not boot with latest firmware.

pfeerick, 2021-09-24:
And if *this* one breaks, I get to blame it on someone else ^^^ :-P

[t18-9bb08c2.zip](https://github.com/EdgeTX/edgetx/files/7225049/t18-9bb08c2.zip)

RipperGOT, 2021-09-24:
Tested in my Radioking and i have the same problem. 

With t18-7fcc3a8, no problem!

But i have the same jitter problem. 😅

pfeerick, 2021-09-24:
Yeah, well, I think we've already established the RK TX18S likes to do the jitterbug, and the factory firmware must be pretty aggressively filtering it out somehow.

Steini63, 2021-09-24:
t18-9bb08c2 fails.

RipperGOT, 2021-09-24:
> Yeah, well, I think we've already established the RK TX18S likes to do the jitterbug, and the factory firmware must be pretty aggressively filtering it out somehow.

Yes yes friend! I was just saying it to inform and eliminate my frustration! 😅
Maybe OpenTX 2.3.14 filter settings could be used. It has a lot less jitter.

RipperGOT, 2021-09-24:
> t18-9bb08c2 fails.

I'll try it tonight! Now i go to work

pfeerick, 2021-09-24:
😢 Well, I have had my amusement in private. So at first glance maybe something in the dynamic MPM proto stuff has buggered things up. Can we get a copy of the models and radio folder, and then we can see if there is something mucked up in the configs. 

Once you've done that, can you try the following - create a new model - use either 9bb08c2 or 2.5.0-rc1, etc... (something that we know crashes when it doesn't like something) then power cycle the radio - it should boot up fine since nothing is turned on. Then turn the internal MPM on, and power cycle it again. Does is successfully power on, or does it freeze on boot?

Steini63, 2021-09-24:
[TX18s.zip](https://github.com/EdgeTX/edgetx/files/7225232/TX18s.zip)

model1 is named "Multi intern" - that is the one that fails on booting
model2 is named "Multi off" - boots o.k
model3 is named "Multi extern" - boots ok

All models are freshly created on Tx (nothing converted or so).

Steini63, 2021-09-24:
> Once you've done that, can you try the following - create a new model - use either 9bb08c2 or 2.5.0-rc1, etc... (something that we know crashes when it doesn't like something) then power cycle the [...]

Did that again with latest 2.6.0. 

1. Create new model - boots up ok with MPD deactivated.
2. Activate Internal RF / MULTI, first entry "FlySky", no other changes - boot fail

Steini63, 2021-09-24:
There is one special thing with t18-9bb08c2:
If I try to change model to "Multi intern" I got an "EMERGENCY MODE" and after reboot the model is not changed to "Multi intern".
On the other problematic firmwares "Multi intern" works.

pfeerick, 2021-09-24:
I'm off to bed now, but thank you for testing that out, and so quickly! :)

Steini63, 2021-09-24:
Thanks for helping so quickly - have to go AFK too.

MervDale, 2021-09-24:
Unfortunately my mates RK TX18S failed again, only workrd for a short while, same issues as above, he has gone back to OTX on this radio, may put ETX on his RM TX16S.   sorry I didn't report it earlier.

raphaelcoeffic, 2021-09-24:
@Steini63 just to exclude everything else: have you tried with a 100% blank SD card? Then create new model, etc... My T18 does not want to block/crash (I have one with touch screen, which is technically the same as the TX18S).

RipperGOT, 2021-09-25:
OK, I have tried again and I still have the error, but it only happens if I change the protocol.

Flash- boot- no changes- reboot ---- ok
Boot- multi intern- default flsky -no more changes - reboot multiple times --- ok
Boot- multi intern- change to any protocol- reboot- screen displayed fails safe not activated - freeze - remove battery - boot ----emergency mode.

Steini63, 2021-09-25:
> @Steini63 just to exclude everything else: have you tried with a 100% blank SD card? Then create new model, etc... My T18 does not want to block/crash (I have one with touch screen, which is technically the same as the TX18S).

Rechecked with another sd-card:
- sd-card formatted
- sd-card prepared with edgetx-flasher (english voicefiles only)
- boot ok
- calibration ok
- reboot ok
- set internal MPM to flysky
- reboot fails

Steini63, 2021-09-27:
Just tried:
Issue still exist in 2.5.0 RC2 and nightly from today.

raphaelcoeffic, 2021-09-27:
> Just tried:
> Issue still exist in 2.5.0 RC2 and nightly from today.

I’m running out of ideas, since I cannot make it crash on my T18…. Does the MPM have all the protocols, or is it maybe one with only CC2500?

raphaelcoeffic, 2021-09-27:
@Steini63 you don’t have a STlinkv2 or anything like that? We need someone to hook a crashing radio to SWD and get a stack trace of the whole thing.

RipperGOT, 2021-09-28:
> @Steini63 you don’t have a STlinkv2 or anything like that? We need someone to hook a crashing radio to SWD and get a stack trace of the whole thing. Worst case we can use desktop sharing to guide you through the process.

Same problem with RC2, freeze on warning trhottle screen.

And sorry, y don´t have a STlink programer.

Steini63, 2021-09-28:
> Does the MPM have all the protocols, or is it maybe one with only CC2500?

No, it's 4in1. Maybe this is is difference - do you have 5in1? 

> @Steini63 you don’t have a STlinkv2 or anything like that? We need someone to hook a crashing radio to SWD and get a stack trace of the whole thing. Worst case we can use desktop sharing to guide you through the process.

I have two different cheapo china St-Link V2 and I'm willing to use them (but have limited experience).

RipperGOT, 2021-09-28:
My RadioKing have the JP5in1

[image]

Steini63, 2021-09-28:
Same here - a 4in1 labeled as JP5IN1.

RipperGOT, 2021-09-28:
> Same here - a 4in1 labeled as JP5IN1. Even the numbers on the bottom right of the board are identical.

Sorry, another question, can you read #702 and say your experience? Thanks friend!

raphaelcoeffic, 2021-09-28:
> No, it's 4in1. Maybe this is is difference - do you have 5in1?
>
>
> I have two different cheapo china St-Link V2 and I'm willing to use them (but have limited experience).

That should be ok, might get dirty however :-D

raphaelcoeffic, 2021-09-28:
@Steini63 please contact me via PM on discord (pafleraf). We will arrange something.

MervDale, 2021-09-29:
Just talking to my mate with the RK TX18S and he has mentioned that when you go to update the MPM firmware for this radio you only get the choice of the RK TX18S  Internal Module (5 in 1)  there is no option of  MPM (4 in 1) firmware is this possibly where the problem lies with the Radios crashing ?

raphaelcoeffic, 2021-09-29:
> Just talking to my mate with the RK TX18S and he has mentioned that when you go to update the MPM firmware for this radio you only get the choice of the RK TX18S Internal Module (5 in 1) there is no option of MPM (4 in 1) firmware is this possibly where the problem lies with the Radios crashing ?

Do you mean the firmware on the MPM is wrong? That still does not mean we should be crashing 😇

pfeerick, 2021-09-29:
I think he just means the MPM is different on the RK TX16S - it's a 5IN1 rather than a 4IN1 - just in case that meant it gives an unexpected response that b0rks things.

raphaelcoeffic, 2021-09-29:
> I think he just means the MPM is different on the RK TX16S - it's a 5IN1 rather than a 4IN1 - just in case that meant it gives an unexpected response that b0rks things. It seemed like on the TX12 if it has the wrong firmware on the MPM it just ignores it, so no harm done there 😆

Nevermind, I'm getting such a crashing radio sent to me.

MervDale, 2021-09-29:
What I was trying to say is that the RK TX18S ONLY has the 4 in 1 MPM   but you can only load the 5 in 1 MPM firmware

pfeerick, 2021-09-29:
I think you mean can only *download* the 5in1 from https://downloads.multi-module.org/ under the RK TX18  ... oh, that is interesting... and it's actually the firmware for the t18 that they offer up. Since AFAIK the matching to a transmitter is only a nicety for hte main part, have you tried loading a TX16S 4in1 MPM firmware, to get actual 4in1 firmware on it? Or just the file filters to specifically pick the STM 4in1 with Serial link?

pfeerick, 2021-09-29:
I've opened an issue to have that dropdown corrected : https://github.com/benlye/multi-downloads/issues/5

MervDale, 2021-09-29:
I assume we know it's the same MPM board  as the STM32 4-1, it does say on the MPM that it is JP5IN1 which as I understand has the 5th protocol removed ?

MervDale, 2021-09-29:
So that was the origin of my question in that we have the later iterations of ETX looking for something that is not there and saying "  Bugger I'm outa here " !  so to speak ;-Q

raphaelcoeffic, 2021-09-29:
> So that was the origin of my question in that we have the later iterations of ETX looking for something that is not there and saying " Bugger I'm outa here " ! so to speak ;-Q

We’ll see, I really don’t know for now.

RipperGOT, 2021-09-29:
> Well, that's just weird... looking at an [internal picture of the RK TX18S](https://shop.makerfire.com/products/radioking-tx18s-2-4g-hall-sensor-gimbal-opentx-remote-controller-type-c-charging-radio), I can see what you mean about it saying it's the JP5IN1, but the 2.4Ghz antenna is connected to the top, [rather than an internal PCB antenna, and presumably the 915/868Mhz sub-board that is on the underside is not installed](https://www.aliexpress.com/item/1005001669492767.html), effectively making it a 4IN1... In other words, I would think that whether they fit the 915/868Mhz or not would determine if it's a 5IN1 or a 4IN1...

Exactly!!! The 915 / 868Mhz sub-board is not installed! It really is a 4IN1
You have been ahead of me with the answer.
I work driving at night and I can't write from the truck! 😅 Safety first !!!

richardclli, 2021-09-30:
> 
> 
> I think you mean can only _download_ the 5in1 from https://downloads.multi-module.org/ under the RK TX18? ... oh, that is interesting... and it's actually the firmware for the t18 that they offer up. Since AFAIK the matching to a transmitter is only a nicety for the main part, have you tried loading a TX16S 4in1 MPM firmware, to get actual 4in1 firmware on it? Or just the file filters to specifically pick the STM 4in1 with Serial link?
> 
> [image]

As I know for the firmware, all T18/TX18S should use the t18int no matter whether it is a 5 in 1 or a 4 in 1.  As I know T16 int module and T18 int module are not interchangable and they need different firmware.  Because the T18 4 in 1 is just 5 in 1 with 900 MHz module removed.

pfeerick, 2021-09-30:
> As I know for the firmware, all T18/TX18S should use the t18int no matter whether it is a 5 in 1 or a 4 in 1. As I know T16 int module and T18 int module are not interchangable and they need different firmware. Because the T18 4 in 1 is just 5 in 1 with 900 MHz module removed.

Just to be 100% clear, before I have to open another issue, you are saying that for the RK TX18S (not the Lite), that the MPM does **not** work with the standard STM32 4-in-1, Serial firmware? And instead needs the module type to be Jumper T18 5-in-1? As it primarily only sets the MULTI_5IN1_INTERNAL flag, which in turn enables/configures the SX1276 868/915 transceiver.

richardclli, 2021-09-30:
> Just to be 100% clear, before I have to open another issue, you are saying that for the RK TX18S (not the Lite), that the MPM does **not** work with the standard STM32 4-in-1, Serial firmware? And [...]

You can check and see, for Jumper there are 2 int modules board, one for T16 and one for T18, I have both a T16 and a T18 before, and noticed that the T18 5 in 1 will not work in T16, and T16 4 in 1 will not work in T18.  Which means the T16 and T18 modules are different (maybe the inv noinv design).

So in the T18 4 in 1 use the same PCB as a T18 5 in 1, it is possible that using the stm-serial version (normally for T16 4 in 1) will cause problem.

So I think that all depends on which PCB it is using.

richardclli, 2021-09-30:
Just check again https://downloads.multi-module.org/

Seems the TX18S internal 4 in 1 used stm-serial but not the t18int.   Which means the module is not using 5 in 1 PCB?  Then the problem is  EdgeTX cannot use T18 to support TX18S internal 4 in 1 directly, something need to be changed.

I think we need to gather more information.

richardclli, 2021-09-30:
Maybe @norulers  can explain more.

pfeerick, 2021-09-30:
> Just check again https://downloads.multi-module.org/

This is because I asked them to update it (as well as add the CC2500 entry for the RK TX18S Lite)... hence why I wanted clarification. 

As I can see that there are two different boards - what I am trying to work out is does the TX18S (**not** the T18) board run the `stm-serial` firmware, or does it need to `stm-5in1` variant, (meaning it must identify at runtime that the SX1276 is missing and thus declare any requests to use the protocols it services as invalid if selected). As if that is the case, I need to open a new request to have the RK TX18S switched back to `stm-5in1`.

closed 2021-10-06 by pfeerick

rotorman, 2021-10-31:
Hardware fix is for RadioKing TX18S that solved the MPM power-up hang is explained here:
https://www.rcgroups.com/forums/showthread.php?3924159-Diagnosing-and-solving-system-hang-on-RadioKing-TX18S-while-turning-on-the-int-MPM

efeksk, 2022-05-22:
Merhaba. Bir Radioking uzaktan kumandam var. Edgetx ürün yazılımını yükledim. Bazı sorunlar yaşadım ve opentx kurmaya karar verdim. Ancak, bellenimi yüklerken taranis x9d bellenimini hatayla parlattım. Uzaktan kumandam şimdi çalışmıyor. bu durumu nasıl düzeltirim. Kumanda hiç açılmıyor. sadece usb'yi bağladığımda bilgisayar tarafından smt32 bootloader görülüyor.

rotorman, 2022-05-23:
@efeksk See https://github.com/EdgeTX/edgetx/wiki/Unbrick-your-radio
For non-developmental questions, please use forums (e.g. RCGroups) or Discord instead in the future.

commit 03188cd3 (2021-12-28): [t18/tx18s] add some delay after turning the internal module on

This allows to avoid power supply instabilities.

Resolves #802