EdgeTX/edgetx #803: Jumper T18 Bootloader Failure 2.5 RC1

Ga1j1n, 2021-09-24:
Jumper T18 Bootloader Failure 2.5 RC1
Freshly updated boot loader / Firmware, Bootloader functions to flash but does not appear to closedown requiring power disconnection, will not boot into 2.5 firmware producing a black screen scenario, again power disconnection required, re-flashing to 2.4 resolves the issue

T16 updated at the same time operates as  expected

May relate to the RadioKing TX18s issue reported earlier as they share hardware #802

raphaelcoeffic, 2021-09-24:
It’s most probably the same issue ;-)

raphaelcoeffic, 2021-09-24:
@Ga1j1n just to exclude everything else: have you tried with a 100% blank SD card? Then create new model, etc... My T18 does not want to block/crash.

raphaelcoeffic, 2021-09-24:
> @Ga1j1n just to exclude everything else: have you tried with a 100% blank SD card? Then create new model, etc... My T18 does not want to block/crash.

Also, are you sure this is a bootloader issue? I mean, if you can start the bootloader itself, it is probably fine.

norulers, 2021-09-25:
This problem occurs when touch support is enabled on devices that do not contain touch hardware
`set(FLAVOUR t18)
    set(LUA_EXPORT lua_export_t16)
    add_definitions(-DRADIO_T18)
    add_definitions(-DRADIO_FAMILY_T16)
    option(INTERNAL_MODULE_MULTI "Support for MULTI internal module" ON)
    option(BLUETOOTH "Support for bluetooth module" OFF)
    set(HARDWARE_TOUCH YES)
    add_definitions(-DMANUFACTURER_JUMPER)`

raphaelcoeffic, 2021-09-25:
@norulers Thx, I'd bet there is something in the init code that will block in case the sensor is not found. I'll check this.

raphaelcoeffic, 2021-09-25:
> @norulers Thx, I'd bet there is something in the init code that will block in case the sensor is not found. I'll check this.

@rotorman checked, and it looks alright. He used a TX16S with the touch sensor disconnected: no issue there.

rotorman, 2021-09-25:
Here TX16S running present main branch code with touch controller disconnected
[image]
:

norulers, 2021-09-25:
[T18V2.zip](https://github.com/EdgeTX/edgetx/files/7229456/T18V2.zip
[T18.zip](https://github.com/EdgeTX/edgetx/files/7229463/T18.zip)

V2 version enabled  set(HARDWARE_TOUCH YES)

Ga1j1n, 2021-09-25:
I just tested the above, t18 completed flash and ALMOST booted to UI and then crashed with speaker noise, T18 V2 would not complete flash, reverted to 2.4 release.

I'd like to be able to activate touch input in the future, I have a spare Tx16s screen awaiting a transplant as the V2 part is not available as a spare no matter where I look

I did not update bootloader just in case, did I need to?

raphaelcoeffic, 2021-09-26:
> I just tested the above, t18 completed flash and ALMOST booted to UI and then crashed with speaker noise, T18 V2 would not complete flash, reverted to 2.4 release.
>
> I'd like to be able to activate [...]

Can you please post pictures of the main board on the T18? As it seems, there should be 2 resistors close to the MCU pulling I2C low.

raphaelcoeffic, 2021-09-26:
@Ga1j1n can you please test the T18 build from #808 ? That should fix the boot issue.

norulers, 2021-09-26:
> @Ga1j1n能否请您测试#808的 T18 版本？那应该可以解决启动问题。

It can be started normally on T18 without touch hardware

closed 2021-09-26 by raphaelcoeffic

Ga1j1n, 2021-09-26:
My T18 is likely a V1 as it is none touch, board photos as requested

[https://photos.app.goo.gl/39ipU5Rqn19X8Foz9](url)

[https://photos.app.goo.gl/QujQF8XLyGARUf2AA](url)

raphaelcoeffic, 2021-09-26:
> My T18 is likely a V1 as it is none touch, board photos as requested
> 
> [https://photos.app.goo.gl/39ipU5Rqn19X8Foz9](url)
> 
> [https://photos.app.goo.gl/QujQF8XLyGARUf2AA](url)

Thx, you should be able to boot now with the latest version from `main`, or the build from #808 (check the artifacts).

norulers, 2021-09-27:
[image]

If you need to add touch hardware, you need to remove the two resistors in the red box

efeksk, 2022-05-22:
Hello. I have a Radioking remote. I have installed the Edgetx firmware. I had some problems and decided to install opentx. However, I flashed the taranis x9d firmware with error while installing the firmware. My remote control is not working now. how do i fix this situation. The remote won't turn on at all. smt32 bootloader is seen by computer only when i connect usb.

rotorman, 2022-05-23:
@efeksk Please do not cross-post the same question in multiple places. I answered you already in: https://github.com/EdgeTX/edgetx/issues/802#issuecomment-1134299465

commit 6a7cb4ae (2021-09-26): [color] Move touch panel init to UI task

This allows for the timeout checks in I2C code to work properly as they are based on SysTick IRQ, which is started by FreeRTOS when the scheduler is started.

Resolves #803
commit aa0ebb26 (2021-09-26): [color] Move touch panel init to UI task

This allows for the timeout checks in I2C code to work properly as they are based on SysTick IRQ, which is started by FreeRTOS when the scheduler is started.

Resolves #803
PR #808 (2021-09-26): [color] Move touch panel init to UI task
This allows for the timeout checks in I2C code to work properly as they are based on SysTick IRQ, which is started by FreeRTOS when the scheduler is started.

Resolves #803
commit 004989ea (2021-09-26): [color] Move touch panel init to UI task

This allows for the timeout checks in I2C code to work properly as they are based on SysTick IRQ, which is started by FreeRTOS when the scheduler is started.

Resolves #803