EdgeTX/edgetx #806: NV14 voltage telemetry

eshifri, 2021-09-25:
**Describe the bug**
NV14 telemetry sensors for RX voltage and model battery voltage have the same name A3 (this does not match OTX naming: A1 for battery).
On the telemetry screen in the model setup both sensors has values (different, of course).
On the widget first A3 (A1 actually) has correct value.  A3 (battery voltage) is showing zero.
SF setup to report A3 as zero as well.

**To Reproduce**
Bind  a model to the NV14. Connect voltage sensor FS-CVT01 to the receiver and to the model battery.
Open Telemetry page and run discover sensors.
Observe two A3 sensors.
Sensor 0 shows RX voltage (~5v).
Sensor 1 shows battery voltage (e.g. for 3 cell battery 11.4V).

Go to widget setup and create a widget to show second A3 sensor.
Observe 0V.

Create an SF to play value for A3.
Value will be reported as "zero" (no units).

**Expected behavior**
RX voltage should be named A1 on the telemetry screen.
Battery voltage should be A3 and should be available for widgets and SFs.

**Screenshots**

**General information**
 - EdgeTX 2
[image]
[image]
.5RC1

 - Hardware NV14

pfeerick, 2021-09-26:
Now that you mention it, it doesn't match the names when a Flysky recevier with built-in telemetry is used with the TX16S and MPM, which I would have thought should be the same for both it and the NV14.

eshifri, 2021-09-26:
Looks like NV14 has its own names in flysky_nv14.cpp.
RQly is also mentioned twice there.
https://github.com/EdgeTX/edgetx/blob/main/radio/src/telemetry/flysky_nv14.cpp#L93

Interestingly enough, changes the names in this array has no effect on displayed names (and values).

pfeerick, 2021-09-26:
Just double check  this for me please... (make sure you **_delete_** the telemetry sensors first, and then run discovery again ;) )

```diff
diff --git a/radio/src/telemetry/flysky_nv14.cpp b/radio/src/telemetry/flysky_nv14.cpp
index 62e392c4a..2a205ba00 100644
--- a/radio/src/telemetry/flysky_nv14.cpp
+++ b/radio/src/telemetry/flysky_nv14.cpp
@@ -92,7 +92,7 @@ union nv14SensorData {
 
 const FlyskyNv14Sensor Nv14Sensor[]=
 {
-    {FLYSKY_FIXED_RX_VOLTAGE,  0, STR_SENSOR_A3,          UNIT_VOLTS,         2, 0, 2, false},
+    {FLYSKY_FIXED_RX_VOLTAGE,  0, STR_SENSOR_A1,          UNIT_VOLTS,         2, 0, 2, false},
     {FLYSKY_SENSOR_RX_SIGNAL,  0, STR_SENSOR_RX_QUALITY,  UNIT_RAW,           0, 0, 2, false},
     {FLYSKY_SENSOR_RX_RSSI,    0, STR_SENSOR_RSSI,        UNIT_DB,            0, 0, 2, true,},
     {FLYSKY_SENSOR_RX_NOISE,   0, STR_SENSOR_RX_NOISE,    UNIT_DB,            0, 0, 2, true},
```

eshifri, 2021-09-26:
I have already tried it: the name on the telemetry setup page changes to A1; name in the sensors selection menu changes to A1 (makes sense :-) ).
A3 is 11.36V on the telemetry setup screen and zeo everywhere else.

pfeerick, 2021-09-26:
Out of curisity, since I don't have a "FS-CVT01" sensor, I created a custom sensor, A3, and it shows up the telemetry as 100.00v ... but values widget is instead showing 1 by default, and showing the position of the elevator stick ... in other words... is yours doing the same - showing a value from somewhere else? 😁

eshifri, 2021-09-26:
I tried to move all controls - it shows reliable and constant zero. -:0)
I have also tried to create a calculated sensor: when A3 is selected as a source (in the menu), it displays zero in the setup screen.

eshifri, 2021-09-26:
Looks like this is common for all external sensors, connecting to RX through **iBUS**: I connected altitude sensor and got zeros for pressure and altitude. On the sensor setup screen pressure and altitude numbers were reasonable. 
On this screen all RX internal telemetry has ID==0. For A3 ID==1; for pressure and altitude ID ==2.
So looks like for ID!=0 there is a problem.

I thought that the problem was in the flySkyNv14ProcessTelemetryPacket(), since it ignores multipacket sensors (if I read it correctly), but porting the code from OpenTX-NV14 did not fix the problem.

eshifri, 2021-09-29:
One more discovery:
when selecting source for widget user gets 3 choices for every input except throttle.
[image]

only the first of them is working - other display zero.
I think this hints that inputs are interpreted as telemetry and the whole indexing system is incorrect.

Interestingly enough for TX16S this does not happen, but user gets a suggestion to select multiple instances of  Input 32 and lots of nameless telemetry signals.

pfeerick, 2021-09-29:
Hm, I'm not seeing this when binding to my FS-X8B on the NV14... no doubled up inputs, and no nameless signals. 

Is this with a model created on the transmitter itself, and no Companion conversion crazyness?

> Interestingly enough for TX16S this does not happen, but user gets a suggestion to select multiple instances of Input 32 and lots of nameless telemetry signals.

Is this specifically when using the same receiver/telemetry sensors? As that would suggest it is specific to the flysky telemetry support. i.e. I'm not seeing anything out of the ordinary on the TX16S with the same FS-X8B - no double ups, no weird telemetry.

eshifri, 2021-09-29:
For NV14 the model was created in companion and cleaned on the TX.
No nameless telemetry signals on NV14.

On TX16S - model converted from OTX 2.3.14. This one does not have duplicate inputs, but lots of nameless signals.
RX and sensors are exactly the same re-bound to one of the radios: FS iA10B, FS CVT01 and FS CAT01.

eshifri, 2021-09-29:
I created a new model on NV14 - it looks better: no duplicated inputs.
A3 shows -1; A3- -100; A3+ 0. Pressure and altitude - zero.

On the telemetry setup page everything is OK.

pfeerick, 2021-09-29:
To check for the naming on the TX16S, would be a good idea to create a new model there and do a telemetry discover, and see if the transmitter mangles it itself, otherwise, we're looking purely at something being mangled between it and Companion. 

A3 is the FS-CVT01 voltage, isn't it? Is it hooked up to an actual battery, or just floating?

eshifri, 2021-09-29:
I will create a new model on TX16S later today (and report the results :-) )
That being said, the existing models on TX16S were direct port from OTX 2.3.14 without Companion involvement.

A3 is a voltage coming from FS CVT01. The sensor is connected to the balancing connector of the battery.

eshifri, 2021-09-29:
On the newly created model on TX16S and the same RX and sensors:
1. Widget setup allows to chose up to 32 inputs and 32 channels; 63 LSs and all GVs (not used by the model).
2. No duplicate entries.
2. Telemetry icon is not available on the left side of the window.
3. A3 is working.
4. Altimeter is working.

On TX16S altimeter provides 3 values: pressure, temperature and altitude (as it should, I believe).
On NV14 temperature is missing (even on the telemetry setup page) and altitude, I suspect, is not working. I had this suspicion for a long time: looks like the altimeter support is not ported from NV14 OTX (it arrived there at a very late stage).

eshifri, 2021-09-29:
Some news. 
I have rebased my code with the last two days of changes and the extra items in selection menu are gone.
Telemetry icon on the left is available.
On both TX16S and NV14.
Complete triumph of the science over inert matter! :-)

Another change that occurred after rebase is on NV14 (at least) every time I run sensor rediscovery they are coming in the different order (if memory serves me it was the same in OTX). This would be OK, but the widgets reference sensors by index - not by name, so telemetry screen is screwed. (Not too bad, but unpleasant.)

The thing that became clear out of it, is that the problem on NV14 is related to the sensor index. E.g. in one of the rediscoveries A1 became the last sensor in the list and it was showing zero in the widget. A3 moved up and it was working.

Altimeter does not work on NV14: pressure looks reasonable and it changes (noise), but altitude is constant and temperature is not available. I know this sensor performs some calculations and its data format is not trivial. Funny enough, it is working correctly with multimodule (on TX16S).

raphaelcoeffic, 2021-09-29:
The way widgets index sensors (and not just widgets, it‘s the same everywhere) is really annoying. In the long term, we need to get rid of this legacy issue.

eshifri, 2021-09-30:
So independently of the order only 6 first sensors are available outside the model_telemetry page.

eshifri, 2021-11-02:
Interestingly enough, lua scripts are fetching correct values.

PR #1211 (2021-12-10): change source_t from uint8_t to uint16_t for NV14. #806
Fixes #806 

Summary of changes: change data type of source_t from uint8_t to uint16_t. 
One byte is not sufficient any more for all sensor sources.
I do not understand why it works for other targets, but apparently it does.
Most likely, almost everything is considered to be PCBFRSKY. :-)
closed 2021-12-10 by pfeerick