EdgeTX/edgetx #809: FrSky X10 Express fw package missing

nostromo123, 2021-09-26:
EdgeTX v2.4.0 lists one x10 fw package, v2.5.0-rc1 and nightly list two fw x10 packages (one with add. 'access'.)

Trying to install one of those on my X10S Express with bl 2.5 or 2.6 here both fw packages (with or without 'access') report  'Wrong PCB detected'.

Would be nice to have a working  x10-express.bin in the fw package.

br KH

raphaelcoeffic, 2021-09-26:
@nostromo123 please use the latest builds from `2.5` and `main` branch, that has been reported to work properly.

closed 2021-09-26 by raphaelcoeffic

nostromo123, 2021-09-26:
I know. I build the fw according to the Ubuntu 20.04 building manual from edgeTX main. This works.

But it should be mentioned, that there is no ready package available for those, who don't want to build ....

br KH

raphaelcoeffic, 2021-09-26:
Will be fixed in RC3 then.

pfeerick, 2021-09-26:
It will be included in RC2, since the fix is already in place in the nighties, so you should actually be able to grab it from the nightly build. But thank you for the report! :)

nostromo123, 2021-09-26:
Thank you. That is, what I wanted to bring forward here.

On that page here https://github.com/EdgeTX/edgetx/releases/tag/v2.4.0

there is this statement

[image]

No problem for the experts here, but a 'normal' user will download the fw.zip attached and be a bit disappointed.

Watched quite some YT videos and the bottom line just is, 'It is so easy to try edgeTX'.

br KH