EdgeTX/edgetx #811: Extending Companion telemetry simulation capabilites

pfeerick, 2021-09-26:
From [RCGroups](https://www.rcgroups.com/forums/showpost.php?p=47868371&postcount=716): 
> Thanks for adding Companion to EdgeTX!!! It is what I was waiting for before trying EdgeTX, and it seems the development team has made a very good start.

> Now that we have Companion, it brings up something that has been concerning me about OTX Companion for quite a while. Telemetry support in OTX Companion simulator has not been updated for quite a while. The telemetry input screen is the same as it was several years ago. It was fine when it first came out because it could simulate all of the Frsky telemetry sensors that were available at the time. When the multi-module was added to the mix, the number of telemetry inputs rose dramatically. OTX developers did a great job of updating the transmitter firmware to make use of the new telemetry options, but Companion simulator lagged behind. It is still not possible to simulate quite a bit of the new telemetry from non-Frsky receivers that can be bound to the multi-module.It is also not possible to simulate the telemetry available from some newer Frsky equipment, for example the Neuron ESC.

> Does EdgeTX have plans to update Companion simulator so that it will handle all of the telemetry available?

stronnag, 2021-09-26:
At the moment (current github master), companion2{5,6} fails to correctly replay even FrSky logs. Only RSSI appears in the simulator from the log, other fields are ignored. OTX replays all the logged telemetry data.

pfeerick, 2021-09-26:
Mentioned this in a new issue as it's a bug report rather than discussion of a feature enhancement. ;)

pfeerick, 2021-09-27:
Further discussion: https://www.rcgroups.com/forums/showthread.php?3916381-Official-EdgeTX-Discussion-Thread/page51#post47888485

dwheeld, 2021-10-08:
Extending Companion telemetry simulation capabilities by making the telemetry simulator compatible with all multimodule telemetry.

More information:

Here is more information about the problem. Some of this will be repeated from before, but I want to get all of my ideas in one message if possible.

First, when companion is simulating a model using any compatible receiver (frsky or non frsky), the sensor ids on the transmitter match those in companion simulator. This proves that when files and settings are downloaded from the transmitter into companion, the correct id's are downloaded also. From this, we know this is NOT where the problem is located. Companion seems to know what the correct id’s are, but the simulator is not using them.
[image]
Here is what I think is causing the problem:

The companion simulators ‘Telemetry Simulator’ page (pic2) only contains the sensors that were available several years back, and they are all for frsky sensors. 

[image]

This became a huge problem when multi-modules came on the scene. Multi modules allow the use of receivers/telemetry that are not listed on the ‘Telemetry Simulator’ page (pic 2). Also, some of the new frsky sensors (neuron esc to name one) telemetry items are not included. The id numbers for the original sensors were always used when companion simulator was used. With all of the new sensors available now, the id numbers are not known by the ‘Telemetry Simulator’ input page. We can't simulate these sensors because the ‘Telemetry Simulator’ page (pic2) does not include them, and it does not seem possible to have one page with all of the possible sensors available for simulation. As an example, I have 1 model with a non-frsky receiver which has 42 telemetry sensors discovered. 37 of these do NOT appear on companion simulators ‘Telemetry Simulator’ input page (pic2)!

How is simulated telemetry handled now?:

It appears that the way things are done now is that companion simulator takes the data (voltage, current, etc.) shown on the ‘Telemetry Simulator’ input page (pic 2), encodes it with an Id number, sends it to the Telemetry page (pic 3) which decodes it by using the id numbers and places the data in the proper place. I don't know why it is done this way. Encoding and then decoding the data seems totally unnecessary to me. The data could be simply placed into the correct spot without encoding/decoding along the way. No id numbers would be necessary. When simulating, we simply want the data being available to test our model programming. 

[image]

The only ways I can think of to correct this is to either (1) delete the Telemetry Simulator input page (pic2) and in its place, show a copy of the simulated Telemetry Page (pic3) docked to one side of the simulated transmitter at all times (pic 4). 

This would be dynamically built each time companion simulator is started, using the same data that builds the Telemetry page. 

[image]

Or (2) completely redo the Telemetry Simulator input page. (pic 2) This would also be dynamically built when called for, but would include the Replay SD Log File choice at the top as well as the Simulate, Pause simulation when hidden., and Set RSSI to zero when paused. boxes at the top (pic5). 

[image]

In either case, (1) or (2) above, I would strip out all of the companion simulator code that does the encoding/decoding of telemetry data, and simply use the data entered  into the  new Telemetry Simulator input page (pic4 or pic5) and place it directly into the simulated Telemetry page (pic 3). (More easily said than done I'm sure!)