EdgeTX/edgetx #812: Companion2{5,6} fails to correctly replay FrSky logs.

pfeerick, 2021-09-26:
At the moment (current github master), companion2{5,6} fails to correctly replay even FrSky logs. Only RSSI appears in the simulator from the log, other fields are ignored. OTX replays all the logged telemetry data.

_Originally posted by @stronnag in https://github.com/EdgeTX/edgetx/issues/811#issuecomment-927283034_

pfeerick, 2021-09-26:
Moved this to a new issue as it's a bug report rather than discussion of a feature enhancement. ;)

jtaylor2, 2021-09-27:
Need more detail about what doesn't work and perhaps a log.  I can replay a openTX log in today's master self compiled with openSUSE 15.3.  See attached screen capture:

https://user-images.githubusercontent.com/6879281/134832256-f13ffd0e-0bba-40d8-a036-9812efe240d9.mp4

stronnag, 2021-09-27:
1. Using 2021-09-27 master  (07d0a663680f) compiled on Arch.
2. Models converted a while ago from OTX 2.3.14 using ETX 2.4.0
3. Model telemetry performs correctly "live" with either OS.
3. Replaying the attached log file [otx-test-log.csv](https://github.com/EdgeTX/edgetx/files/7235390/otx-test-log.csv)
* In ETX, very few sensors are recognised; rediscovering the sensors in the simulator shows them with different IDs; after that all extant widgets then show the wrong sensor.
* In OTX, all sensors are seen as expected and display in widgets, it is not necessary to rediscover sensors.

There's a couple of matroska videos in the tar file (showing ETX and OTX behaviours [replays.tar.gz](https://github.com/EdgeTX/edgetx/files/7235733/replays.tar.gz) )

stronnag, 2021-09-27:
Created a **new** model in companion. Discovered sensors. Telemetry works (which was the whole point of the exercise).
[image]

stronnag, 2021-09-27:
As far as I'm concerned, this can be closed.

pfeerick, 2021-09-27:
True, telemetry works, but the underlying issue - basically some sort of conversion corruption is still there, right?

stronnag, 2021-09-27:
Yes, it would appear that there is an underlying issue of conversion for some telemetry sensors, as the new model works in ETX simulator.

closed 2023-02-25 by gagarinlg