EdgeTX/edgetx #814: Logical switches are messed when copying model between NV14 and companion

eshifri, 2021-09-26:
**Describe the bug**
I'm using not the latest build of companion, so maybe it was fixed...
I created a new model in companion for NV14.
Setup LSs and SFs.
Copied the model to the radio.
It is difficult describe what exactly happens, but lots of LS were remapped to higher (no-defined) LS. E.g. LS04 became LS16.)

I fixed everything and copied the model from the radio back to companion - everything was wrong again: some LSs (with higher numbers) were re-mapped to inputs.

**To Reproduce**
see above.
**Expected behavior**
Exact copy of the model setup. :-)

**General information**
 - EdgeTX version current code for TX and companion 2.5
 - Hardware NV14 and Windows 10

elecpower, 2021-09-26:
It is highly likely it is not just LS that are 'messed' up and be evident across switches and sources.

elecpower, 2021-09-28:
I've got it

PR #832 (2021-09-30): Fix shift in NV14 switches during Companion import and export
Fixes #814
Cause NV14 has no multi position pots
Companion and Simulator testing:
- no shifts for new models created using TS16S, X9D+ and NV14 profiles
- using NV14 profile importing and converting TS16S models and settings
- using X9D+ profile importing and converting NV14 models and settings
closed 2021-09-30 by pfeerick