EdgeTX/edgetx #815: Companion Installer does not include sdl.dll

jmxp69, 2021-09-27:
git branch main
clean build environment from scratch following wiki
cmake -G "MSYS Makefiles" -DCMAKE_PREFIX_PATH=$HOME/5.12.9/mingw73_64 -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DFLYSKY_HALL_STICKS=YES -DCMAKE_BUILD_TYPE=Release ../ &&

make -j`nproc` companion &&
strip companion.exe &&
make -j`nproc` simulator &&
strip simulator.exe &&
make -j`nproc` libsimulator &&
make installer
[image]

jmxp69, 2021-09-27:
sdl.dll can be manually copied from \msys64\mingw64\bin to \program files\edgetx\companion 2.6\ and it works
[image]
.

pfeerick, 2021-09-27:
That's not the 'companion installer' ... that would be the installer you run to actually install companion without needing to compile it yourself ;)

Have you added "C:\msys64\mingw64\bin" to you user environment path?   I'm not sure why that's not in the wiki guide anymore, I'm sure it used to be? @rotorman

jmxp69, 2021-09-27:
I'm not sure I understand what you're saying. 

When I follow the process in the wiki to compile the installer "make installer" I'm left with an executable that when run, installs Companion, but there is a DLL required by the resultant shortcut that precludes Companion from running until I manually put sdl.dll into the companion directory. If that installer is given to another person that doesn't have msys64 on their system, it likewise fails and they don't have the msys64 dll to copy.

Shouldn't companion-windows-2.6.0.exe which comes from make installer yield an executable that installs companion that can be run without error?

jmxp69, 2021-09-27:
~/edgetx/edgetx_main/build-output
rm -rf *

There is no companion-windows-2.6.0.exe at this time.
cmake -G "MSYS Makefiles" -DCMAKE_PREFIX_PATH=$HOME/5.12.9/mingw73_64 -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DFLYSKY_HALL_STICKS=YES -DCMAKE_BUILD_TYPE=Release ../

make -j`nproc` companion 
There is no companion-windows-2.6.0.exe at this time.

make -j`nproc` simulator
There is no companion-windows-2.6.0.exe at this time.

make -j`nproc` libsimulator
There is no companion-windows-2.6.0.exe at this time.

make installer
09/27/2021  01:51           104,481 companion-windows-2.6.0.exe

I don't think it's appropriate to ask anybody to change their environment path in order to use the installer being made from the steps on the wiki. This should be self contained.

jmxp69, 2021-09-27:
> That's not the 'companion installer' ... that would be the installer you run to actually install companion without needing to compile it yourself ;)
>
> Have you added "C:\msys64\mingw64\bin" to you [...]

The error message is the result of running the companion shortcut installed in the menu after running the companion installer
companion-windows-2.6.0.exe

rburrow87, 2021-09-27:
Yeah I also find that if I `make companion` and `make installer` and install it, I get that error about `SDL.DLL` missing. It appears to be there when I look in the Companion 2.5 folder from RC1.

raphaelcoeffic, 2021-09-27:
just add the following to your cmake line: `-DSDL_LIBRARY_PATH=/mingw64/bin/`.

closed 2021-09-27 by raphaelcoeffic

jmxp69, 2021-09-27:
> just add the following to your cmake line: `-DSDL_LIBRARY_PATH=/mingw64/bin/`. I guess you ignored the big warning about SDL not being found ;-)

I watch the data go scrolling by, but there are lots of messages. I didn't see anything about SDL not being found until after the install completed and it asks to launch companion. 
[code]
Install the project...
-- Install configuration: "Release"
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/companion.exe
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/simulator.exe
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/libopentx-tx16s-simulator.dll
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/dfu-util.exe
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/libusb0.dll
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/libusb-1.0.dll
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/license.txt
C:\msys64\home\jmhop\edgetx\edgetx_main\build-output\companion.exe 64 bit, release executable
Adding Qt5Svg for qsvgicon.dll
Direct dependencies: Qt5Core Qt5Gui Qt5Multimedia Qt5Network Qt5PrintSupport Qt5Widgets
All dependencies   : Qt5Core Qt5Gui Qt5Multimedia Qt5Network Qt5PrintSupport Qt5Widgets
To be deployed     : Qt5Core Qt5Gui Qt5Multimedia Qt5Network Qt5PrintSupport Qt5Svg Qt5Widgets
Updating Qt5Core.dll.
Updating Qt5Gui.dll.
Updating Qt5Multimedia.dll.
Updating Qt5Network.dll.
Updating Qt5PrintSupport.dll.
Updating Qt5Svg.dll.
Updating Qt5Widgets.dll.
Updating libgcc_s_seh-1.dll.
Updating libstdc++-6.dll.
Updating libwinpthread-1.dll.
Patching Qt5Core.dll...
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/audio.
Updating qtaudio_windows.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/bearer.
Updating qgenericbearer.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/iconengines.
Updating qsvgicon.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/imageformats.
Updating qgif.dll.
Updating qicns.dll.
Updating qico.dll.
Updating qjpeg.dll.
Updating qsvg.dll.
Updating qtga.dll.
Updating qtiff.dll.
Updating qwbmp.dll.
Updating qwebp.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/mediaservice.
Updating dsengine.dll.
Updating qtmedia_audioengine.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/platforms.
Updating qwindows.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/playlistformats.
Updating qtmultimedia_m3u.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/printsupport.
Updating windowsprintersupport.dll.
Creating directory C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/styles.
Updating qwindowsvistastyle.dll.
Processing config: C:\msys64\mingw64\etc\nsisconf.nsh
Processing script file: "C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/companion/companion.nsi" (ACP)

Processed 1 file, writing output (amd64-unicode):

Output: "C:\msys64\home\jmhop\edgetx\edgetx_main\build-output\companion\companion-windows-2.6.0.exe"
Install: 7 pages (448 bytes), 1 section (16408 bytes), 604 instructions (16912 bytes), 1903 strings (18942 bytes), 2 language tables (772 bytes).
Uninstall: 3 pages (256 bytes), 2 sections (32816 bytes), 154 instructions (4312 bytes), 1038 strings (8268 bytes), 2 language tables (564 bytes).
[/code]

In any event, adding the above to cmake resolved it. Thanks.

Install the project...
-- Install configuration: "Release"
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/companion.exe
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/simulator.exe
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/libopentx-tx16s-simulator.dll
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/dfu-util.exe
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/libusb0.dll
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/libusb-1.0.dll
-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/license.txt
**-- Installing: C:/msys64/home/jmhop/edgetx/edgetx_main/build-output/Release/SDL.dll**
C:\msys64\home\jmhop\edgetx\edgetx_main\build-output\companion.exe 64 bit, release executable

raphaelcoeffic, 2021-09-27:
@jmxp69 you should see it when executing `cmake` the first time.

rotorman, 2021-09-27:
Thank you for the feedback! Just updated the instructions in Wiki to include -DSDL_LIBRARY_PATH=/mingw64/bin/
https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Windows-10-(MSYS2)