EdgeTX/edgetx #817: Add possibility to set logical switch via LUA-script

wimalopaan, 2021-09-27:
Sometime it would be really handy to have a logical switch (LS) that could be set / reset via a lua-function.

The logical switch should get a "function" of kind: LUA.

Only for these logical switches a lua.setLS(id) or lua.resetLS(id) should have an effect.

JimB40, 2021-09-27:
Can you give example of this sometime?

wimalopaan, 2021-09-27:
The most obvious example would be to have the possibility to defined UI-buttons and UI-switches in LUA-widgets, which act the same way as other switches.
Actually I am doing something similar in my own widgets. but this would make it more usable.

jfrickmann, 2021-10-03:
You can set a Global Variable from Lua and use that in  Logical Switch. Or you can user a Custom/Mixer Script and set a logical switch based on the output value from that.

wimalopaan, 2021-10-03:
Thank you.
That is actually the way I do that. But both ways look weird to me. 

If we propose the GV -> LS way, then we should increase the number of GVs, say to 64, at least for the colorLCDs (may the memory of the monoLCDs is too small).

I was a bit unclear in saying LUA-script. But I really meant mostly LUA-widgets. So the mixerscript -> LS way is not possible for widgets.

Actually I increase the number of GVs in my own compilations, but that unfortunately makes the storage format incompatible.

jfrickmann, 2021-11-10:
How about a Lua function that can set a sticky LS?
`setStickySwitch(idx, value)`, 
where `value` is boolean, and if the switch selected by `idx` is not sticky, then nothing happens.

jfrickmann, 2021-11-10:
@wimalopaan, please check PR #1019!

wimalopaan, 2021-11-10:
Thanks!
Looks good to me.
But I would like to change it also for non-colorlcd radios. I don't think that the 16 int's hurt so much for the smaller radios.

wimalopaan, 2021-11-10:
Thanks!,

Looks good to me!

BTW: actually I'm using a workaround by modifying a LS from

{"a>x", MAX, 0} --> true

to

{"a=x", MAX, 0} --> false

jfrickmann, 2021-11-15:
@wimalopaan - please check #1056!

wimalopaan, 2021-11-15:
Thanks!
Will do.

closed 2021-11-29 by pfeerick