EdgeTX/edgetx #821: There is no way to exit full screen widget on the RX without HW keys. (NV14)

eshifri, 2021-09-27:
**Describe the bug**
Double tap on the widget brings it to full screen.
Touch screen does not provide an option to exit full screen without HW keys. (I know there is a work around, but it is not clean.)
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/colorlcd/widget.cpp#L77

We need either handle long touch or use scroll touch (not exactly consistent).

**To Reproduce**
see above

**Expected behavior**
Long press should close the full screen mode.

**General information**
 - EdgeTX 2.5RC1
 - Hardware NV14

raphaelcoeffic, 2021-09-27:
But the left trim switch pressed long does work, right?

eshifri, 2021-09-27:
Yes, of course. This is what I meant by "workaround".

pfeerick, 2021-09-27:
> Long press should close the full screen mode.

I'm not sure I can go along with this yet, as pressing the HW-key left trim is not a "workaround" and is likely to be a permanent thing. As will being able to navigate lus apps via one of the trim joysticks, just like with the otx fork for NV14. 

I'm also concerned about introducing different behaviours just for one transmitter - I think the UX should be as consistent as possible across them all. So what exactly would this look like to the end user, for all colourLCD radios with and without touch?

eshifri, 2021-09-27:
When pressing the trim button, there is a chance to change trim, that is why I'm not sure this can be the permanent final solution.
I completely agree that behavior should be the same for all color lcd RXs with touch screen, so solution should not be for NV14 only. Currently forTX16S there is also no way to exit full screen with touch.
How about using double tap for both enter and exit? This is easy to implement.

pfeerick, 2021-09-28:
Lets see what @JimB40 and @jfrickmann think... since they are our resident UI/UX and Lua guru's. 

I don't have any objections to that, but also feel that means widgets loose the option to get double tab events... perhaps it does need to be long press... i.e. double tap to open, long press three seconds to exit... (yes, it would need to be long to make sure it's not confused with a press and hold event if lua widgets can use that?).

rotorman, 2021-09-28:
With FS PL18 we might have in near future another "special" radio with no real physical UI buttons for interaction.
I would prefer a touch option to achieve the exit instead of re-purposing other buttons. Even, if that would mean holding the finger for ca. 3 seconds down or similar. Although the details are best to be worked out by UI/UX experts than me, who has lim->0 knowledge about this.

pfeerick, 2021-09-28:
Wash your mouth out... unintelligent comments indeed! 😁 That is an excellent point, if no buttons for UI are present, we definitely need a "touch-only" option... it's just a matter of figuring out the best option...

raphaelcoeffic, 2021-09-28:
BTW, the PL-18 does not have any trim joysticks, and thus no "top-button" (just the normal trims), so that it might be an issue to keep the same tactic with that radio later.

jfrickmann, 2021-09-28:
I believe that is what `lcd.exitFullScreen()` is for. Of course, it is on the script writer to implement it in the UI.
https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/lcd-functions-less-than-greater-than-luadoc-begin-lcd/lcd.exitfullscreen

JimB40, 2021-09-28:
Comment is okay. We may consider assigning one of touch gestures ie pinch to execute make widget exit from app mode.
Once TouchUI API is extended.

jfrickmann, 2021-09-28:
Here is a demo:
https://youtu.be/2UESfTJWemg

rotorman, 2021-09-28:
Nice! When I compare the video buttons to typical Desktop GUIs, then I see two differences:
a) typically "X" symbol is used for close (instead of up arrow for "back")
b) and rightmost button would be the close/back and minimize to the left of it.
I guess it would make sense to keep uniform with other GUIs (I looked here at Win10 and Ubuntu with Gnome as examples).

jfrickmann, 2021-09-28:
The way I look at it, is "minimize" is when we go back to widget mode. I.e. the widget screen is the task bar. You cannot close a widget app completely, so no "X".
The other button with the up arrow is "go back to previous screen", so it is a little different.
Smells a little like Windows, but not the same!

eshifri, 2021-09-28:
This is very nice indeed, but what about a situation when there is no lua script associated with widget?
Used double-tap on it just out of curiosity and now just needs to return back to main screen?

jfrickmann, 2021-09-28:
Maybe something like Assistive Touch on an iPhone?

eshifri, 2021-09-28:
This will take real estate on the screen, won't it? There used to be a ">" button there that was removed.

jfrickmann, 2021-09-28:
> This will take real estate on the screen, won't it? There used to be a ">" button there that was removed.

Yes. But the button can be dragged around so the user can move it out of the way