EdgeTX/edgetx #822: Companion: Linux buii

howels, 2021-09-27:
OpenTX supplies Linux builds for companion as DEB packages, please also offer these for EdgeTX companion.

akabaev, 2021-09-28:
Please do not limit yourself to .deb only. Using flatpack or appimage to package the app and it dependencies will make it much more useful for many more Linux users.

howels, 2021-09-28:
> I think when we pursue this I we should be looking at AppImage as it provides the best self-contained package IMO... .deb is just unnecessarily limiting to debian based distros... Yes, there are options like `debtap` for Arch, but better to not go down that avenue if at all possible since we have the chance to basically start again ;)

Sure, AppImage is a good choice if you are creating a new build system. I thought it was more a case of re-enabling the DEB output in existing Companion build routines but AppImage is a great solution.

pfeerick, 2021-09-28:
For starters, sure, but it's not a long term solution.

closed 2021-10-06 by pfeerick

howels, 2021-10-06:
If anyone is looking for the Linux binary it's here for this PR: https://github.com/EdgeTX/edgetx/suites/3921596471/artifacts/98082355

stronnag, 2021-10-06:
Please can we have `deb` as well, at least that's usable on modern distros (even not Debian).

```
$ ninja package
[0/1] Run CPack packaging tool...
CPack: Create package using External
CPack: Install projects
CPack: - Install project: EdgeTX []
CPack: Create package
ninja: error: unknown target 'DESTDIR=/home/jrh/Projects/edgetx_main/build/_CPack_Packages/Linux/External/AppImage'
Simulator plugins: 
exe_list_cmd = 
linuxdeployqt 7 (commit b469748), build 734 built on 2021-06-13 13:17:57 UTC
ERROR: The host system is too new.
Please run on a system with a glibc version no newer than what comes with the oldest
currently still-supported mainstream distribution (Ubuntu Bionic), which is glibc 2.27.
This is so that the resulting bundle will work on most still-supported Linux distributions.
For more information, please see
https://github.com/probonopd/linuxdeployqt/issues/340
CPack: - package: /home/jrh/Projects/edgetx_main/build/companion26-0.1.1-Linux.json generated.
```

Back to the home made tarball script for local machines ...

raphaelcoeffic, 2021-10-06:
> Please can we have `deb` as well, at least that's usable on modern distros (even not Debian).
>
> ```
> $ ninja package
> [0/1] Run CPack packaging tool...
> CPack: Create package using External
> CPack: [...]

You can do 2 things:
- `make install`
- or compile with the docker container, which has everything to make the proper AppImage.

I cannot support Debian packages, as it does not really make sense for fast moving software.

stronnag, 2021-10-06:
I only wish to compile it once locally  (and install it), and then install a "package" to two other local machines. Deb worked, appimage doesn't. A trival tarball does too. I see no reason for the complexity of docker for my use case, thank you.

raphaelcoeffic, 2021-10-07:
> One other thing @raphaelcoeffic is that the AppImage should have +x set by default, the current build doesn't and so cannot be directly executed.

It does for me when built locally. I suspect GitHub to modify stuff when they package it as a ZIP file. The AppImage Website describes how to make it executable pretty nicely, so this seems to be a generic way of distributing these packages.