EdgeTX/edgetx #823: Swiping between screens slow

unluckyfrog, 2021-09-27:
**Describe the bug**
*Swiping to inav lua very slow and not smooth.
*inav lua colours not reproduced correctly.
*When scrolling back to main screen top bar colour has changed.

**To Reproduce**
update to 2.5.0 rc1 and or 2.5.0 rc2 (tx16s-994c4ff.bin)

**Expected behavior**
Using inav lua should not change colour on main screen top bar.
Swiping should be smooth(it was in 2.4)
inav lua should work. It does in 2.4 but in 2.5 the fonts are all incorrect.
Re-flashing to 2.4 solves the issue.

**Screenshots**

https://user-images.githubusercontent.com/57021696/134995517-f35eb9b9-e0d6-4404-a3b6-69c5337d9944.mp4

**General information**
 - EdgeTX version 2.5.0 rc 1/2
 - Hardware Tx16s

pfeerick, 2021-09-28:
Looks like something went wrong with the video upload.

Top bar colour is a known issue, and requires changes in the iNav lua script to not set colours the way it does. Please check you are running the _current_ version of hte widget, as the PR to resolve this was merged yesterday.

unluckyfrog, 2021-09-28:
> Update: I've just installed it myself as I use the iNav lua with on of my
> models, and it does fix the issue with messing with system theme colour
> settings and inav colours. The slow scrolling when using touch may just
> have to be something we live with for now... there is just a minor delay if
> you use the hardware page scroll buttons when switching to it, but it
> doesn't have any slide glitches.

Yep just uploaded new version of inav lua and working well. Still slow
scroll but pg buttons work. Thanks

JimB40, 2021-09-28:
If I may suggest. Use own independent color table instead of ETX constants. Inav uses own constant colors independent of OS system colors anyway. So it's much better to define them as variables holding 24bit hex 0xXXXXXX values or proper color flags inside script.
We still work on ETX on UI so color constant names and assignment to UI elements may change.

stronnag, 2021-09-28:
Thanks for the suggestion; I started on that path, and due no doubt to my inexperience with the project and lua ended up with issues between ETX and OTX. The current approach was pragmatic to get something working on both platforms. 

I'll maybe revisit it. The whole inav widget could benefit from significant re-factoring in any case.

JimB40, 2021-09-28:
If you want keep it as flags (faster)

local COLORS = {
  BACK = lcd.RGB(0x000000),
  GRAY = lcd.RGB(0x808080),
  RED  = lcd.RGB(0xFF0000),
}

lcd.drawText(0,0,"Sample Midsized Gray Text", COLORS.GRAY+MIDSIZE)
-- sample red rectangle
lcd.drawFilledRectangle(10,10,100,200,COLORS.RED)

this technique is not cross-compatible with OTX

stronnag, 2021-09-28:
> this technique is not cross-compatible with OTX

Unfortunately, we need to remain compatible with OTX. 
Maybe we'll end up with a colour table approach that abuses / restores system flags / theme for OTX and behaves nicely for ETX.

JimB40, 2021-09-28:
OTX needs to set this CUSTOM_COLOR every time so I used proxy function that returns CUSTOM_COLOR with proper color if on OTX or proper color flag value if on ETX. And to simplify you can use metatable to swap colors' table access behavior from value to function output if on OTX

stronnag, 2021-09-28:
That's pretty much where the inav widget is going as well.

stronnag, 2021-09-29:
WIP, but much improved scrolling speed, shouldn't touch the system theme on Edge
https://github.com/stronnag/inav-EdgeTX-OpenTX-Telemetry-Widget/releases/tag/2.0.1

unluckyfrog, 2021-09-29:
Just loaded. Awesome thx

pfeerick, 2021-10-01:
Oh my... just tried it out... I think the speed is back to normal, so it's just the animation now :)

stronnag, 2021-10-01:
We've merged https://github.com/iNavFlight/OpenTX-Telemetry-Widget/pull/13 into the inav  lua  widget master, so we're done for now.
@pfeerick please elaborate on "it's just the animation now".
You can probably close this and add any further comments on the inav lua github.

pfeerick, 2021-10-01:
@stronnag Sure... "just the animation now" as in... the only delay is the actual swipe animation... I don't see any real delay that I could attribute to the Lua script anymore. Thank you for this, it looks great, and I'll be properly testing it out tomorrow when flying :)

closed 2021-10-01 by pfeerick