EdgeTX/edgetx #825: Make it possible to switch the whole widget-page with all NxM widgets to fullscreen (touch-aware)

wimalopaan, 2021-09-28:
I just came across the following limitation: 

if one uses e.g. the 4x1 widget layout it would sometime be preferable to set the whole combination of all 4 widgets to fullscreen-mode. Primarily because all these widgets want to receice touch-events. 

Sure one can implement this via a manager lua-script, but this seems to be sort of the wrong place and too much boilerplate in lua.

pfeerick, 2021-09-28:
Is this somewhat of an overlap with #712 ? 

I'm not quite sure what you mean by switching all four widgets to full-screen mode... as you can only have one widget fullscreen at a time!?!

wimalopaan, 2021-09-28:
Up to now you can only switch one widget to fullscreen, say one out of four in a 4x1 layout.

What I mean is to switch the whole 4x1-screen to fullscreen, so that

* the 4x1-layout with all the four widgets occupies the full LCD, and
* that all widgets receive touch events

JimB40, 2021-09-28:
Hehe that is when communication suffers mixing under one name full-screen OS layout with widget app-mode (working in full-screen). I'd say "I've told you" but well I just repeat to introduce and "Widget App-Mode" name and do some corrections in doc before it grows :)

philmoz, 2024-02-17:
How does the system know which widget a key press should go to?

Touch screen could work out which widget; but key presses might be a problem.

pfeerick, 2024-02-17:
For key input, maybe concept of which widget is in focus? i.e. border/frame around the widget signifies which ones receives key input, which is either the last one touched, or the one selected using the roller + enter navigation. I suspect this is why there was no mention of key input in the request :laughing:

pfeerick, 2024-02-17:
Correct. Not many other choices though for hard key navigation. i.e. just like entering a model name is a lot harder with hard keys than touch. So with hard keys only you would have to RTN to get out of the widget, scroll to the next, and long press to get back into it (if I'm understanding the model correctly). I'm sure this would increase the sales of touch screens if people want to use this capability and it be easier. The alternative being to reserve a hard key for switching which widget is in focus, but then which key/event, and how loudly will widget authors scream about losing (another) key/event.

philmoz, 2024-02-17:
It may also be less of an issue once widgets support Lvgl controls. 
The widgets could have buttons and controls on them which would then be selectable with the roller (without entering 'app mode').