EdgeTX/edgetx #830: bluetooth support tx16s

sgsracer256, 2021-09-28:
hi, i added a bluetooth module, can it be activated in the firmware ?

pfeerick, 2021-09-28:
You need to build a custom firmware and add the 'BLUETOOTH=Y' option.

IdefixRC, 2021-11-14:
> hi, i added a bluetooth module, can it be activated in the firmware ?

I compiled it for my own use. Might be helpful for you as well.
(options used: -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=YES -DLUA=YES -DBLUETOOTH=YES -DCMAKE_BUILD_TYPE=Release)
NOTE: Fresh from the compiler. Have yet to test it on my TX16

https://drive.google.com/file/d/1cYBqwtYQhXBDUsYgOQqagFYhhTAKVMUA/view?usp=sharing

Luznatural, 2022-07-31:
Does anybody have a compilation for TX16S with bluetooth on Edgetx 2.7.1?

pfeerick, 2022-08-01:
There's several builds for TX16S with bluetooth enabled in the by request for #1951

closed 2022-08-01 by pfeerick