EdgeTX/edgetx #834: Simulator under Win shows UTC time

rotorman, 2021-09-29:
Simulator does not show local time under Windows:
[image]

@elecpower would you have an idea how to fix this? Thx!

raphaelcoeffic, 2021-09-29:
Currently the code uses „localtime()“, but it would probably better work to use the window API:
https://docs.microsoft.com/en-us/windows/win32/api/sysinfoapi/nf-sysinfoapi-getlocaltime

pfeerick, 2021-09-30:
Simulator on Ubuntu is UTC for me also. I'm not sure this is actually using `localtime` as that should be returning a correct local time, as opposted to `gmtime()` ... are we talking about `RTC_GetTime()`  in `simpgmspace.cpp`, as it doesn't even appear to be being used.

raphaelcoeffic, 2021-09-30:
> Simulator/simu on Ubuntu is UTC for me also. I'm not sure this is actually using `localtime` as that should be returning a correct local time, as opposted to `gmtime()` ... are we talking about `RTC_GetTime()` in `simpgmspace.cpp`, as it doesn't even appear to be being used?

Oh! I blindly assumed this is what is being used! Where is the time coming from then?

pfeerick, 2021-09-30:
Not a clue... I set a static time, no effect, and even shoved a `exit(0)` to make it simply halt, and it just carried in...  (`simu`, in this case...
)

elecpower, 2021-09-30:
When I get back to my computer I'll point you to the code.

raphaelcoeffic, 2021-09-30:
> When I get back to my computer I'll point you to the code.

I think this is what is used:
https://github.com/EdgeTX/edgetx/blob/36ec9356f3f9d8a03ab51a45fed01579b0f40ea4/radio/src/targets/simu/simpgmspace.cpp#L214

raphaelcoeffic, 2021-09-30:
@pfeerick sorry, I went for the simpler solution, especially to be able to use the existing RTC functions in `rtc.c` (see #838).

pfeerick, 2021-09-30:
> @pfeerick sorry, I went for the simpler solution, especially to be able to use the existing RTC functions in `rtc.c` (see #838).

Why sorry, did you not read this above? 😆 

> Now you just need to improve it :-P

Anyway, it looks like `gmktime` was the magic sauce here... I was originally using the same code as you ended up with, but mktime, and it was returning UTC time, so ended up with the functions I found on stackexchange.

PR #838 (2021-10-01): Simu: Adjust RTC according to local time on start
Resolves #834
closed 2021-10-01 by pfeerick