EdgeTX/edgetx #836: NV14 Internal Module protocol

pfeerick, 2021-09-30:
@elecpower Just noticed that for the NV14 it's showing AFHDS3 for the internal module protocol instead of AFHDS2A

Also, whilst I beleive the RF output power can be changed, that isn't exposed in the settings atm, but I don't know which side is right as far as the bin file settings (i.e. simply missing from interface, or Companion shouldn't be showing it). 

[image]

elecpower, 2021-09-30:
I haven't tackled the supported modules and protocols yet as until the NV14 sim started working with touch I had no interface to see the radio gui. I haven't checked the code recently but I have a feeling the internal protocol isn't even known to Companion so will need the whole shebang.
Anyway I'll start on a PR.
Keep checking as I'm sure there are other settings that will need adding/tweaking.

pfeerick, 2021-09-30:
It "seems" to enable and disable it ok as it is ... I haven't tried changing the options yet, so we'll see what happens. I just suspected it was going to be more than changing the text in the dropdown, so better to let the expert look at it! 😆

PR #3783 (2023-09-07): chores(flysky): separate AFHDS2A and AFHDS3 into 2 different module type
Fixes #836

This should allow to remove some of the hacks added earlier.
closed 2023-09-07 by pfeerick