EdgeTX/edgetx #841: enter in statistics wont reset statistics

ishay320, 2021-10-01:
**Describe the bug**
in the statistics menu the "[enter] to reset" is not working, not when pressed on the screen and not when press on the enter button

**To Reproduce**
navigate to statistics(main screen press enter and choose statistics)
press enter and it wont reset

**Expected behavior**
when press enter it should reset the statistics

**Screenshots**
[image]

**General information**
 - EdgeTX version "Dauntless" 2.5.0-RC2
 - Hardware tx16s

pfeerick, 2021-10-01:
Partly needs a wording change "Press to reset", but yes, generally still a bug 😁

ishay320, 2021-10-03:
I'm trying to learn how to help, so have I found the line which the button setup is?
https://github.com/EdgeTX/edgetx/blob/0cbb14bc11aeae89bd11a6740d79fda73a6acea4/radio/src/gui/colorlcd/view_statistics.cpp#L138-L144

pfeerick, 2021-10-04:
Two changes are needed IMO... the translation string for STR_MENUTORESET need changing (after verifying that it's not used somehwere else where it would be the wrong text to show) and also a handler added so the long ENT also works there - as on non-touch colour lcd there is no way to get to that button AFAICT from a check on the radio itself.

ishay320, 2021-10-04:
I have checked and the translation show in all radio but with the same syntax
https://github.com/EdgeTX/edgetx/blob/0cbb14bc11aeae89bd11a6740d79fda73a6acea4/radio/src/translations/en.h.txt#L613
I change some stuff, now I want to compile, what's the command for tx16s?

raphaelcoeffic, 2021-10-05:
Guys, did you verify the button is on focus?

pfeerick, 2021-10-06:
That would be the rather unhelpful `NO_FOCUS` that can be removed if you'd like to do the honours @ishay320 ;)

https://github.com/EdgeTX/edgetx/blob/0cbb14bc11aeae89bd11a6740d79fda73a6acea4/radio/src/gui/colorlcd/view_statistics.cpp#L144

https://github.com/EdgeTX/edgetx/blob/0cbb14bc11aeae89bd11a6740d79fda73a6acea4/radio/src/gui/colorlcd/view_statistics.cpp#L238

ishay320, 2021-10-11:
The throttle value , throttle %  and curve is not resetting, 
I have fixed it, tell me if the function in class ThrottleCurveWindow  -  resetCurve is on par (line 49).

`

`

commit 247d2b0f (2021-10-06): Statistics page theming + ENTER fix (#872)

* Statistics screen theming
Basically none of the text was using theme colours.

* Formatting

* Use COLOR_THEME_PRIMARY2 for prefixes

* Remove NO_FOCUS stopping reset button from working
Fixes #841
closed 2021-10-13 by pfeerick