EdgeTX/edgetx #843: Companion not showing all images available for use as model image

pfeerick, 2021-10-01:
@elecpower I have a doozy for you... and this is either a user error, or this bug is also present in OTX as well... 

In a nutshell, I was just creating a new model (new OTX file, TX16S profile), and have the SD card folder selected in Companion settings, and instead of letting me see all 12 `.png` images, it only lets me see three of them. 😮 Gimme my other 9! 😆 

[image]

This happened with the same dataset on both OTX and ETX companion, and with the ETX companion on windows and linux via the test appimage build. 

Attached is the contents of my images folder at the time of the screenshot (just in case it is related to the files/and better mimic). I noticed that if I added a bunch of `*.jpg` files - probably stock ones from OTX, more files were listed, but even then, it wasn't a complete list. 
[IMAGES.zip](https://github.com/EdgeTX/edgetx/files/7267376/IMAGES.zip)

Possibly related (or if not, was the reason I encountered this) is this case where the filename was corrupted, and this users file list was also suitably bare... ([discord comment](https://discord.com/channels/839849772864503828/839849772864503832/893466610898202624))

[image]

[image]

elecpower, 2021-10-01:
Found it and why not resolved in OTX but need to check why the current logic as there could be a hidden gotcha.

MervDale, 2021-10-01:
Sorry to upset the apple cart but I am not having any issues with Companion at all in this regard. 
[image]

pfeerick, 2021-10-02:
You'll only notice it (in my experience of this so far) with small data sets, or if you actually tally up the list of all the images in a directory with those shown.

The fact Neil didn't immediately say "you're dreamin', mate" suggests there really is a bug, but thanks for the feedback

PR #848 (2021-10-05): Fix companion model edit image names
Fixes #843
Edited models in Companion and Simulator checking file names up to 10 chars excl .bmp extn for eeprom and 14 chars inc extn for sd card radios. Tested:
- TX16S
- X9D+

Image list sorted case insensitive to match radio.
Use firmware capabilities rather than specifying boards.
closed 2021-10-05 by pfeerick