EdgeTX/edgetx #847: Build Instructions under Windows 10 (MSYS2) not working

dwheeld, 2021-10-02:
**Describe the bug**

QT dependencies not being installed.

**To Reproduce**
Steps to reproduce the behavior:

Follow the step by step installation in the web page https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Windows-10-%28MSYS2%29

At step:

 $ aqt install 5.12.9 windows desktop win64_mingw73

The last few lines will read:

Arguments: `['C:\\msys64\\mingw64\\bin\\aqt', 'install', '5.12.9', 'windows', 'desktop', 'win64_mingw73']` Host: `uname_result(system='Windows', node='DESKTOP-GLQ5UIR', release='10', version='10.0.19041', machine='AMD64')`

Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.9/multiprocessing/connection.py", line 317, in _recv_bytes

After step:
 make installer

The last few lines will read:

make[3]: *** [companion/src/CMakeFiles/installer.dir/build.make:72: companion/src/CMakeFiles/installer] Error 1
make[2]: *** [CMakeFiles/Makefile2:2231: companion/src/CMakeFiles/installer.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2238: companion/src/CMakeFiles/installer.dir/rule] Error 2
make: *** [Makefile:1034: installer] Error 2

The companion install file is not created.

pfeerick, 2021-10-03:
So how far into this does it fail for you?

And just a double check, you are running this from "MSYS2 MinGW 64-bit (blue icon)", not the purple icon? As you switch over from one to the other part way through. 

```
pfeer@DESKTOP-RSREF72 MINGW64 ~
$ aqt install 5.12.9 windows desktop win64_mingw73
aqtinstall(aqt) v1.2.5 on Python 3.9.6 [CPython GCC 10.3.0 64 bit (AMD64)]
Downloading qtbase...
Downloading qttranslations...
Downloading qtsvg...
Downloading qtactiveqt...
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qttranslations-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 7.84858010
Downloading qtgraphicaleffects...
Finished installation of qtsvg-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 9.43410250
Downloading qtdeclarative...
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtgraphicaleffects-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 6.31923100
Downloading qtwinextras...
Finished installation of qtactiveqt-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 14.82040530
Downloading qtmultimedia...
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtwinextras-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 17.20881710
Downloading qtimageformats...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtimageformats-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 12.56555870
Downloading qttools...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtdeclarative-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 79.43914080
Downloading qtsensors...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtmultimedia-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 84.56134690
Downloading qtwebchannel...
Finished installation of qtsensors-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 13.24427790
Downloading qtwebsockets...
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtwebchannel-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 7.68062110
Downloading qtlocation...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtwebsockets-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 9.71541630
Downloading qtserialport...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtserialport-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 6.86502390
Downloading qtquickcontrols...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtquickcontrols-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 24.86549200
Downloading qtquickcontrols2...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qttools-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 156.50852030
Downloading qtxmlpatterns...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtquickcontrols2-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 114.52052650
Downloading qt3d...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtxmlpatterns-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 65.79100910
Downloading qtcanvas3d...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtlocation-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 170.39971650
Downloading qtscxml...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtcanvas3d-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 20.60788240
Downloading qtserialbus...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtscxml-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 17.64794280
Downloading qtgamepad...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtserialbus-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 15.11098400
Downloading qtspeech...
Finished installation of qtgamepad-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 8.07072740
Downloading x86_64...
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of x86_64-7.3.0-release-posix-seh-rt_v5-rev0-runtime.7z in 5.66365730
Downloading d3dcompiler_47...
Finished installation of qtspeech-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 7.34688850
Downloading qtremoteobjects...
Redirected: ftp.yz.yamagata-u.ac.jp
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of d3dcompiler_47-x64.7z in 7.83446570
Downloading opengl32sw...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtremoteobjects-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 15.97215530
Finished installation of opengl32sw-64-mesa_12_0_rc2.7z in 17.51515080
Finished installation of qtbase-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 371.57479180
Downloading qtconnectivity...
Redirected: ftp.yz.yamagata-u.ac.jp
Finished installation of qtconnectivity-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 22.04793320
Finished installation of qt3d-Windows-Windows_10-Mingw73-Windows-Windows_10-X86_64.7z in 175.08107360
Patching C:/msys64/home/pfeer/5.12.9/mingw73_64/bin/qmake.exe
Patching C:/msys64/home/pfeer/5.12.9/mingw73_64/bin/Qt5Cored.dll
Patching C:/msys64/home/pfeer/5.12.9/mingw73_64/bin/Qt5Core.dll
Finished installation
Time elapsed: 438.28710130 second
```

rotorman, 2021-10-03:
@dwheeld Something went wrong with setting up the build environment. Easiest just to start from scratch. Uninstall MSYS2, delete "C:/msys64/", reboot and start over. Does it work now?

dwheeld, 2021-10-03:
pfeerick 

Yes, I use blue and purple icons as directed. I have tried it on 2 computers. On 1 it fails after issuing pacman -Su. A google search showed that this may be because that computer does not  have enough memory to perform this task. Something about 'threading', which I do not understand. On the other computer it makes it past this fine but fails after make installer is issued, with the messages :

make[3]: *** [companion/src/CMakeFiles/installer.dir/build.make:72: companion/src/CMakeFiles/installer] Error 1
make[2]: *** [CMakeFiles/Makefile2:2231: companion/src/CMakeFiles/installer.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2238: companion/src/CMakeFiles/installer.dir/rule] Error 2
make: *** [Makefile:1034: installer] Error 2

NOTE: It builds companion.exe, in the build-output folder, which executes without a problem after I copied all of the necessary dll files to the same folder. 

-----------------------------------------------------------------

rotorman

I have tried 2 times now, but hate to give up. Will do as you suggest and try again.

rotorman, 2021-10-03:
@dwheeld if you join in EdgeTX Discord, I can try to help you in screen-sharing session (ping me up Risto #5128 in Discord if you are interested)

dwheeld, 2021-10-03:
rotorman

Tried again on both computers with same results.

I appreciate your offer of help! Thank you very much!!! I will definitely take you up on it sometime in the coming weeks. For now, since I am able to build a working companion.exe, I will use that to test and help debug some changes I am going to try in companion.  

I am going make an attempt to work on issue 811. I have been frustrated by this shortcoming in OTX companion for a long time and this is my chance to try to give back. After I get qt installed and working, I will definitely need help figuring out how companion simulator does things now before I can make the necessary changes. I will join EdgeTX Discord at that time and will most likely bug you guys quite a bit....

pfeerick, 2021-10-04:
The earlier `connection.py", line 317, in _recv_bytes` error could have even been a weird sort/not very well handled download server error... as I see our own build system automation had a minor hissy fit whilst trying to do the `aqt` install, but the task was fine when re-run a little while later.

dwheeld, 2021-10-05:
After my last reinstalls in an attempt to get the Creating an installer step to work, which failed, I am now no longer able to build a working companion.exe. I cleared everything out and tried a new install at which time I received multiple error messages, and at one point msys reported I had discovered a bug and please make a bug report. After doing so, msys reported back that they were looking at the bug report, but the errors indicated I don't have enough memory to use msys on my computer. 

I then attempted to follow the steps in Building and debugging with Qt Creator, and that failed also. I asked questions on a qt forum and was informed that it appears to them that my computer will not handle the latest qtcreator. 

So it appears that if I want to work on EdgeTX code, I will have to purchase a new computer. That isn't possible at this time so for now I will not be able to work on issue 811 as I had hoped. 

Thanks for the offers of help and good luck to all.

rotorman, 2021-10-05:
@dwheeld Just to rule out that you were not informed wrongly about your computer - what CPU and how much RAM and how much free space does your hard drive have?
I am assuming you are using Windows, but what Windows version exactly are you using (e.g. I am looking for info in the vain of "Windows 10 Professional 64-bit in version 21H1")?

raphaelcoeffic, 2021-10-05:
Just FYI: I did all the tests and initial dev for this using a VM using only 4GB of RAM, which is kind of a minimum for Windows 10. The VM comes from the Microsoft Windows10 trial program.

dwheeld, 2021-10-05:
System information:

[system information.txt](https://github.com/EdgeTX/edgetx/files/7286775/system.information.txt)

raphaelcoeffic, 2021-10-05:
> System information:
> 
> [system information.txt](https://github.com/EdgeTX/edgetx/files/7286775/system.information.txt)

It might be indeed a bit on the edge. I had 4GB as well, but I switched most things OFF (especially virus detection and anything like security features) to speed things up. I believe Windows10 runs much better with 8GB of memory anyway. You should really think about updating, or, alternatively, use Linux instead (dual boot in your case, as you have really few memory). You need to know that Windows is a bit "waisty" on RAM... Linux or macOS are much more efficient with RAM.

rotorman, 2021-10-05:
@dwheeld After looking through the specs you gathered, although your system is not new, I do not see it being an issue. I extract 4 core Core i7-960 with 4 GB RAM with 378GB free on C:, and Windows 10 64-bit, which all looks perfectly fine to me.
There must be some other issue you are having. My offer still stands that we could try it with screen sharing. In what timezone are you in?

dwheeld, 2021-10-05:
raphaelcoeffic

I would say a little over the edge ;) . One of the install steps, I don't remember which one, takes 3 and a half hours to complete!. I am using a cable modem for internet access, so that should not be a bottleneck. It takes over an  hour and a half to build companion (make -j`nproc` companion).

rotorman 

I have asked a friend to do the install on his computer. He is an avid gamer and his gaming computer is loaded with everything! I will let him do it with me not around to influence him in any way. If it works for him, I will then try it on the same computer. I will also let him do the install on my computer. This should tell me if it is me (most likely) or not. Will let you know how it comes out.