EdgeTX/edgetx #849: Move telemetry to separate task

eshifri, 2021-10-03:
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

It is a question/suggestion since I'm not familiar with all details...
In NV14 OTX @qba667 separated telemetry from mixers into a separate task:
https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/blob/pulses23/radio/src/tasks_arm.cpp#L140
https://www.rcgroups.com/forums/showpost.php?p=45357535

So mixers have higher priority than telemetry.
There are also some provisions there to separate bluetooth and some other components.

Would it make sense to port this code/idea?

raphaelcoeffic, 2021-10-03:
@eshifri this looks appealing, however I would bundle this kind of change together with the module driver re-write. No need to port anything, it is really simple enough. I'd prefer to have my own soup at this place 😇

closed 2022-10-01 by eshifri

raphaelcoeffic, 2022-10-01:
For reference: telemetry has been moved to FreeRTOS timer routine, which uses less resources than a separate task.