EdgeTX/edgetx #850: Keyboard input on Text Widget Partially Obscures Text Field

MartinHugh, 2021-10-03:
Radiomaster TX16S
Edge TX 2.5 RC1

When editing a Text Widget, the setting dialogue box is shown centrally on the screen.  If the Text field is edited,, although the keyboard is shown at the bottom of the screen, it partially obscures (~25% or so) the bottom of the text field making it less than ideal for editing.

Why not show the dialogue at the top of the screen, or even consider moving it there when the keyboard is displayed.

pfeerick, 2021-10-03:
Excellent point, and yes, something will need to be done to improve that. At the present time, widget options are a bit of a pain generally with the on screen keyboard, but it will improve over time.

Runsys16, 2021-10-05:
Hi,
for example, I suggest
For all edit text, the screen keyboard is displayed on the entire screen.
But the first line of this screen is a copy of edit text.

MervDale, 2021-10-06:
Not sure everyone would be aware of this and as a point of interest  everything that is a above the Keyboard is scrollable so it is easy enough to scroll the Text Widget so you can see what your are typing :-)

MervDale, 2021-10-06:
Gee and  I thought the Dev's had done a great job with the keyboard and the widgets , text is the only one that the keyboard impinges on and the made it scrollable, my hat's off to them ;-Q

MartinHugh, 2021-10-06:
I agree, an excellent job, especially given the timescales.  Note that all that an over-sensitive reaction to constructive feedback does, is discourage people from testing, and trying to help improve the product ;-)

pfeerick, 2021-10-06:
> It is not great user experience

You're right, but until it is fixed, a workaround has been offered. And it will be needed, as I don't think there is a quick-fix we can put into effect before 2.5.0 gets released. I feel that it did actually scroll itself up before, but doesn't now because it caused a major glitch the way it was implemented (i.e. locking the UI up completely wasn't seen as a positive to being able to see the text field). 
:laughing: 

I see no problem with either of your suggestions from a end-user perspective, and thank you suggestions on how to improve things rather than just point out imperfect behaviour ;)

I feel that overall, what @Runsys16 proposed is the better option, as it gets away from any screen trickery, and also allows for something else to happen, which is make the keyboard bigger, thus easier to use. I'm not saying this is the way it will go, but that is what I would pick unless some better gets suggested ;)

MartinHugh, 2021-10-06:
Yes I think @Runsys16 's suggestion is good.

gagarinlg, 2023-03-18:
This is fixed since all pages are converted to LVGL. If someone finds a page where it does not work correctly, a new issue is to be opened.

closed 2023-03-18 by gagarinlg