EdgeTX/edgetx #854: Print model only shows headings in ODT files

stronnag, 2021-10-03:
# Describe the bug

When a model is "printed", the output formats behave as:
* ODT: Only headings are shown, no useful data
* PDF. HTML : Model data is shown

## To Reproduce
* Select Print from the model list
* Select ODT as the output format
* View the output file in LibreOffice or other `odt` viewer
* Note the missing data

## Expected behavior

The ODT file should show the same model data as PDF and HTML
[x.odt](https://github.com/EdgeTX/edgetx/files/7274209/x.odt)
[x.pdf](https://github.com/EdgeTX/edgetx/files/7274211/x.pdf)

## See also

[RCG discussion](https://www.rcgroups.com/forums/showpost.php?p=47924645&postcount=6612). Looks like a legacy bug from OpenTX.

stronnag, 2021-10-03:
The `content.xml` file includes the expected data. However, Libreoffice does not display it; according to the RCG report, neither does MS Word. `pandoc` is not interested as there's no `styles.xml`. `abiword` just hangs on it.

Please consider just dropping `odt` and provide **markdown** instead. Or neither when we get `yaml` models.

elecpower, 2021-10-03:
Confirmed option does not produce usable output, has likely never worked and therefore should not have been listed.
However there appears to be a conversion possible but requires minimum Qt version 5.14 to convert from html (print generation) to markdown

PR #859 (2021-10-05): Remove odt file format option from print model
Fixes #854
Fixes in the sense it removes a file format that does not convert the html document generated by the print function with any degree of accuracy
closed 2021-10-05 by pfeerick