EdgeTX/edgetx #858: FrSky S8R or S6R receiver won't update firmware using Edge firmware 2.5.0

Woofie1010, 2021-10-04:
When binding a known and working FrSky S6R or S8R to a new Radiomaster TX16S running Edge 2.5.0 and Mutli protocol firmware 1.3.3.0. The receiver would bind but after binding the RSSI value was 12 and the RSSI warning track was was talking. The green light on the receiver was on but dim. A servo plugged into the receiver would move but in jerks not smooth.

I tried it on an older Radiomaster TX16S running the same Edge tx and multi protocol firmware and it worked fine. So I tried it on a second new Radiomaster TX16S running Open tx firmware and multi protocol version 1.3.2.77 and it had the same low RSSI as the first new Radiomaster radio. Looks as if something is different from the older Radiomaster radios to the new Radiomaster radios.

I was on with John from RC Video Reviews and he helped me troubleshoot the problem. It was decided that the firmware on the receiver needed updated so we setup to use the Radiomaster running Edge 2.5.0 firmware to flash the receiver with the new firmware. No matter what we tried we kept getting the same error message "Firmware update error Device not responding". We tried three different versions of firmware and always the same error. We double checked the cable and connections and still the same error. We could not get the receiver to flash the new firmware. So next we tried a radio running Open tx firmware. The receiver successfully flashed the receiver first time. After the receiver had updated firmware version 2.1.2 it showed RSSI of 104 after binding on the new Radiomaster running Edge tx. The green light on the receiver was a nice solid green and the servo movement was smooth.

pfeerick, 2021-10-04:
Well, that's just annoying... as e2a37f3f7dd10bed5e2e6cb3417e6aaaf92b79ff from 7 days ago (i.e. in 2.5.0-RC2 onwards) fixed this for me on the TX16S for both my XJT module and X8R/D48R receivers...

Woofie1010, 2021-10-04:
Thanks for the reply. After reading your reply I downloaded the latest main version of Edge Tx and installed it. It did indeed fix the issue. The Radiomaster running the latest version of Edge Tx had no issues flashing the S8R receiver with the latest firmware.

After flashing the the firmware (version 2.1.2 for the S8R) all the low RSSI problems are gone.

Thank you.

closed 2021-10-04 by Woofie1010