EdgeTX/edgetx #860: Concrete these undesirable artefact with gauge widget.

stronnag, 2021-10-04:
New theme (concrete) shows an undesirable artefact with gauge widget.
The text is somewhat unreadable using the Concrete theme. It is with OK the other new themes. See image (default v. Concrete).
[image]

JimB40, 2021-10-04:
Widgets are still to be adjusted to use properly etx color contants

philmoz, 2023-10-05:
The gauge uses a function to invert the part of the bar corresponding to the input position.

The problem is that the invert function assumes that the background color for the bar (COLOR_THEME_PRIMARY2) is white.
If the theme changes this color then the invert function misbehaves.

PR #4164 (2023-10-16): fix(color): Gauge invertRect assumed WHITE rather than PRIMARY2
Fixes #860 

Use values of PRIMARY2 color instead of fixed white when inverting the rectangle.

Note: this will break if Gauge widget is ever changed to draw the bar background using a color other than PRIMARY2 (e.g. if the bar background color ever gets added as a widget option).
closed 2023-10-16 by pfeerick