EdgeTX/edgetx #863: NV14 is not booting when flashed with last nightly #155d55d

JimB40, 2021-10-05:
Used this build
https://github.com/EdgeTX/edgetx/actions/runs/1306099483

pfeerick, 2021-10-05:
This is probably my fault... I didn't test #852 on the NV14 before merge, and the firmware for that PR does look like it bricks on the NV14!!! Sorry boss! @raphaelcoeffic

raphaelcoeffic, 2021-10-05:
> This is probably my fault... I didn't test #852 on the NV14 before merge, and the firmware for that PR does look like it bricks on the NV14!!! Sorry boss! @raphaelcoeffic
> 
> nv14-6c74af9.bin (immediately prior commit) works nv14-bfaf529.bin (PR#852) fails to start nv14-ba03d02.bin (merge commit) fails to start

I must admit I did not test on NV14 either..... My fault!

PR #867 (2021-10-06): [nv14] Boot issue fixes
Fixed:
- delay shortened.
- adjusted priority of IRQs used before FreeRTOS is started.

Resolves #863
commit c4dd1c78 (2021-12-28): [nv14] Boot issue fixes

Fixed:
- delay shortened.
- adjusted priority of IRQs used before FreeRTOS is started.

Resolves #863
closed 2021-10-06 by pfeerick