EdgeTX/edgetx #868: ADC filter adds micro steps that mess it up

ctzsnooze, 2021-10-06:
Hi... I'm a Betaflight developer, many of you know me.

I've been dealing with issues arising from people accidentally having the ADC filter active, and have been trying to improve how our flight control code can best deal with the steppy nature of ADC filtered data.

Let me say right from the start that this filter must be off for any drone related flying application.  It should only be active when servos are being used.

And I should point out that I think this bug exists in OpenTx also.

My understanding is that the intent of the ADC filter is to convert a slow stick input to a series of regular steps, to minimised jitter with servo motors.  The height of the step and the period between steps relates to how quickly the sticks are moved.

In the ideal performance of such a filter, the output would be a series of duplicate packets, with occasional larger steps up to a new value.  

The derivative of such a data stream would return strings of zeroes - indicating the same value, with no change from packet to packet - and then an abrupt step to a new value, which would be sent repeatedly, with no change, and again a derivative of zero, until the next step change.

That's how I think it is intended to be.

And a casual look at the data does show a series of steps that appear widely spaced.

But when we look at the derivative, we find that it is not uncommon for a small, single packet height step to be transmitted just before the main step up.  

For example, this graphic shows RC Command at the top, and the derivative, or change amount per step, when EdgeTx is used with ELRS 500hz.  Although the RC command appears to be 'flat with steps', closer examination shows that the flat parts actually have small steps within them.  If the 'plane' had a digital servo, it would 'chatter' with these small steps.  We can see that they are all of identical height.  The larger spikes in derivative indicate the main, larger steps.

What we should see is dead flat between the steps, and a dead flat Derivative between spikes.

The bug is that we don't get that.

[image]

ctzsnooze, 2021-10-06:
Essentially this is an issue for any user of OpenTx who has enabled the ADC filter.

For some reason it is not working as well as it could, and should, be.

This will cause excessive jitter and wear on a digital servo.  It also contributes to issues for drone pilots when the ADC filter is accidentally left on.

For people with drones and firmware like Betaflight, the ADC must always be turned off.  We need smooth continuous data to work with.  Steps and spikes in feedforward are useless to us, they just make the motors hot.

I've improved the way our code handles the situation where a Betaflight user accidentally leaves the ADC on.  

However these small transients impact our ability to determine whether a 'step' is a real pilot input or an ADC type spike.  

It would be great if the post-ADC filter data was actually truly flat in-between steps.

It would also be great if the ADC filter could be stored 'per-model' so that a user can turn it off on all their quadrotor type machines.  There is never any situation where a drone pilot would want the ADC on.

rotorman, 2021-10-07:
@dlktdr For output, I just used on TX16S USART3/AUX1 DEBUG out via FTDI to PC. You can use TRACE() commands. Debug output via USB CDC should work too, with it no FTDI necessary (I did not test that though).

ctzsnooze, 2021-10-08:
It seems to me that the way the ADC filter is intended, is to reduce servo jitter, by algorithmically ensuring that there are no frequent small adjustments.

There are two basic issues in this, I guess.

The first is that the 'raw' incoming gimbal data should be smoothed to the point where it carries some minimal amount of jitter with the least possible delay.

A 1-AUD filter or something like that is unquestionably far superior to a simple lowpass in that it can strongly alleviate jitter when sticks are moving slowly but have minimal delay when they move quickly.  The downside is a little bit of overshoot when a sharp move stops.

However in principle the settings for this filter depend greatly on the nature and magnitude of the noise in the incoming gimbal data.

Most users would want, and benefit from, an adjustable, high-performance gimbal filter.

The second issue is about how to avoid servo jitter.  This has no relevance whatsoever for direct motor control applications like drones.  But it does have relevance when the output device is a servo.

In some applications there are a combination of direct controls and servos, eg when directly controlling a brushless motor in an electric plane.

Therefore it makes more sense that converting a smooth signal into a series of steps - ie making the output 'servo-friendly' - should be done **per-channel.**  Whereas at the moment this is done globally.

From the servo perspective, all that is really needed is an algorithm that, after the gimbal data has been smoothed, limits the steps to some amount, eg 1% of full stick travel, perhaps with a little bit of hysteresis.  The user can specify the step size, and when set to 0, there is no stepping.  And this can be set independently on the *output channel* side of the code, *per-channel*.

The process would be:

- globally smooth the gimbal data (with a parameter for the amount of smoothing, from none to solid)
- optionally, and *per-channel*, make 'servo-friendly' steps in the output (user specifies step size)

Since the 'servo-friendly steps' were done algorithmically, they would be absolutely reliable and clean under all circumstances.  If there was inadequate input smoothing, or simply terrible gimbals, the user might want to increase the step size (decrease resolution).

Anyway, those are my thoughts on this - and again, thanks for looking into it.

raphaelcoeffic, 2021-10-09:
@ctzsnooze I might be misunderstanding the issue, but considering that folks who use servos will also use a lot more mixers, wouldn't it make sense to have the servo jitter filter at the output of the mixer, i.e. directly on the final channels values rather than on the gimbal inputs? Good digital servos have a dead band of around 1us (in the analog PWM), which would be what we should consider as the minimum effecting a change. Considering the mapping (about 1000us to 2000us), that would be mean to smooth based on a 10bit value, right?

dlktdr, 2021-10-09:
If the noise on the analog inputs is filtered enough which to equate to less than <0.5us of jitter, all the resulting steps(mixes etc.) should produce a nice smooth output. In reality this might be a problem on some radios and cause a bit of lag to make it happen, but not too much can be done about that. If the inputs come in smooth, I can't see any reason to reduce the resolution to a servo (with stepping). 

Looking at the code it looks like it was kind of a patch to make a moving average filter work to reduce the noise, which just caused the steps as a result. It's simple to calculate with little cpu + ints. For most applications you would never even notice, I didn't. With a better filter bit more processor/ram and good settings I think we could achieve a really nice result with no further processing down the line.

If Off/Norm/High wouldn't be enough for most radios. Could make it 4 choices. Normal being just enough to reduce most noise but not affect any flight controllers. High for servo system sensitive to jitter, Extreme for sensitive servos or a really bad radio noise? 

I will try your 1AUD filter too. I'll see about making a option to select various filters and values that can be tested. We can do some peer review on before picking a filter and the various constants.

tchleb, 2022-01-10:
> What about switching the filter from the modified moving average based to a 1euro filter. http://cristal.univ-lille.fr/~casiez/1euro/InteractiveDemo/ https://hal.inria.fr/hal-00670496/document
> 
> Worked great for filtering the analog on my NanoBLE which was also very noisy, and doesn't add a ton of lag. Little harder to get the values right. If interested I can add the option.
> 
> I implemented it as a test if want to try it out. https://github.com/dlktdr/edgetx/tree/Analog1EuroFilter set "ONEEURO_ANALOG_FILTER": "YES", will only work on targets with a FPU.
> 
> Filter values are adjustable in the GUI. Not a final solution just a test. Need to find a way to graph the output to compare. [image]
> 
> Also need the approximate frequency update rate of the analog reads to be entered in Frequency (Not sure just guessed at 500Hz :)

I tried to compile your repo with the 1 Euro Filter. Unformatly i'm not able to compile it.
Do you have an idea what I'm doing wrong? 

`git clone --recursive -b 1EuroTestingOutput https://github.com/dlktdr/edgetx.git edgetx-v2.4-1Euro

cd ~/edgetx/edgetx-v2.4-1Euro/

export PATH="~/edgetx/edgetx-v2.4-1Euro/radio/util:$PATH"

mkdir build-tx16s

cd build-tx16s

cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -ONEEURO_ANALOG_FILTER=YES -DCMAKE_BUILD_TYPE=Release ../

make -j`nproc` firmware

....
[ 83%] Building CXX object radio/src/CMakeFiles/firmware.dir/gui/colorlcd/radio_hardware.cpp.o
[ 83%] Building CXX object radio/src/CMakeFiles/firmware.dir/gui/colorlcd/radio_diagkeys.cpp.o
In file included from /home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/libopenui.h:55,
                 from /home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/opentx.h:42,
                 from /home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/filters.h:4,
                 from /home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:26:
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp: In lambda function:
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:329:79: error: 'debugTimers' was not declared in this scope
  329 |       StaticNumber *mt = new StaticNumber(this, grid.getFieldSlot(),GET_VALUE(debugTimers[debugTimerFilter].getLast()), REFRESH_ALWAYS);
      |                                                                               ^~~~~~~~~~~
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/getset_helpers.h:25:62: note: in definition of macro 'GET_VALUE'
   25 | #define GET_VALUE(value)                        [=] { return value; }
      |                                                              ^~~~~
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:329:91: error: 'debugTimerFilter' was not declared in this scope; did you mean 'debugfilter'?
  329 |       StaticNumber *mt = new StaticNumber(this, grid.getFieldSlot(),GET_VALUE(debugTimers[debugTimerFilter].getLast()), REFRESH_ALWAYS);
      |                                                                                           ^~~~~~~~~~~~~~~~
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/getset_helpers.h:25:62: note: in definition of macro 'GET_VALUE'
   25 | #define GET_VALUE(value)                        [=] { return value; }
      |                                                              ^~~~~
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp: In member function 'void FilterWindow::update()':
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:329:135: error: no matching function for call to 'StaticNumber::StaticNumber(FilterWindow*, rect_t, FilterWindow::update()::<lambda()>, const WindowFlags&)'
  329 |       StaticNumber *mt = new StaticNumber(this, grid.getFieldSlot(),GET_VALUE(debugTimers[debugTimerFilter].getLast()), REFRESH_ALWAYS);
      |                                                                                                                                       ^
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:218:5: note: candidate: 'StaticNumber::StaticNumber(Window*, const rect_t&, std::function<int()>, WindowFlags, LcdFlags)'
  218 |     StaticNumber(Window * parent, const rect_t & rect, std::function<int()> getValue, WindowFlags windowFlags = 0, LcdFlags textFlags = 0) :
      |     ^~~~~~~~~~~~
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:218:77: note:   no known conversion for argument 3 from 'FilterWindow::update()::<lambda()>' to 'std::function<int()>'
  218 |     StaticNumber(Window * parent, const rect_t & rect, std::function<int()> getValue, WindowFlags windowFlags = 0, LcdFlags textFlags = 0) :
      |                                                        ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:215:7: note: candidate: 'StaticNumber::StaticNumber(const StaticNumber&)'
  215 | class StaticNumber : public StaticText
      |       ^~~~~~~~~~~~
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:215:7: note:   candidate expects 1 argument, 4 provided
/home/tobias/edgetx/edgetx-v2.4-1Euro/radio/src/gui/colorlcd/radio_hardware.cpp:215:7: note: candidate: 'StaticNumber::StaticNumber(StaticNumber&&)'
[... cut ...]
[ 83%] Building CXX object radio/src/CMakeFiles/firmware.dir/gui/colorlcd/radio_diaganas.cpp.o
make[3]: *** [radio/src/CMakeFiles/firmware.dir/build.make:2754: radio/src/CMakeFiles/firmware.dir/gui/colorlcd/radio_hardware.cpp.o] Fehler 1
make[3]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[2]: *** [CMakeFiles/Makefile2:693: radio/src/CMakeFiles/firmware.dir/all] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:700: radio/src/CMakeFiles/firmware.dir/rule] Fehler 2
make: *** [Makefile:251: firmware] Fehler 2`

pfeerick, 2022-01-10:
Try adding a D in front of `-ONEEURO_ANALOG_FILTER` ... i.e. `-DONEEURO_ANALOG_FILTER=Y`

tchleb, 2022-01-10:
> Try adding a D in front of `-ONEEURO_ANALOG_FILTER` ... i.e. `-DONEEURO_ANALOG_FILTER=Y`

Right I missed the D. I tried again with `-DONEEURO_ANALOG_FILTER=Y` Unfortunately the fault is equal.

JBKingdon, 2022-01-10:
I'd recommend Chris' 1AUD over 1euro. We started with 1euro and it was great but not quite right for gimbal data. 1AUD is the development of 1euro tuned for exactly this use case.

JBKingdon, 2022-01-10:
If you have fp hardware I'd recommend starting with the float version. The fixed point version can be a pain to get the scaling factors right. On esp32 the float version is about the same speed as the fixed point, but I needed a fixed point implementation for gd32v which doesn't have fp hardware.

tchleb, 2022-01-10:
> I had -DDEBUG_TIMERS=y on too. I never got the 1AUD working on there but most of the code is in place. I would probably try implementing the integer based 1AUD mentioned earlier in the comments instead. Got side tracked on other issues. 
> it was in this branch https://github.com/dlktdr/edgetx/tree/1EuroTestingOutput
> 

With `-DDEBUG_TIMERS=YES` I was able to compile. Thank you for your help.

closed 2023-09-05 by raphaelcoeffic