EdgeTX/edgetx #874: Shutdown animation screen

pfeerick, 2021-10-06:
Following on from the discussion in #862, we need to come to a consensus as to what will happen with the shutdown screen, as this is a community driven project, so is driven by what the majority want. 

My proposal was that it use theme colours, since that is what should be determining **all** colours used on the radio. Namely, that `COLOR_THEME_SECONDARY1` would be used for the background, and `COLOR_THEME_PRIMARY1` would be used for the animation icon. 

I felt it was too close to 2.5.0-RC3 which should be the final RC, for this to be given the time it deserves to be seen and discussed, so it has been pushed back to 2.6.0. So over to the floor.

NemesisXB, 2021-11-08:
I agree with using the theme colors. That black and white shutdown animation looks "terrible".

rotorman, 2021-11-08:
We need some thumbs up/thumbs down voting.
How about following instructions - put a thumbs up to first post if you would favor having theme colors as shutdown colors or put thumbs down emoji if you prefer to keep it black and white (as presently).
If you favor a third option, add a comment explaining what that third option would look like.

MervDale, 2021-11-08:
It happens so quickly, does any one really worry about this? which takes up the least resources B/W or colour?

pfeerick, 2021-11-08:
@MervDale That really doesn't matter. This is literally a one line change which was made just before 2.5 release and pulled because of a comment out consistent system UI, and it was too late in the cycle to get feedback. I was actually going to put this up on discord this week to get the crowd to resolve it once and for all, majority rules ;) I still will, and may get into a habit of putting *something* up every couple of weeks for community engagement ;)

PR #1182 (2021-11-29): Use theme colours for shutdown animation/screen
Resolves #874

Summary of changes:
- add `isDefaultTheme()` bool to ThemePersistance()
- Use COLOR_THEME_SECONDARY1 as background, COLOR_THEME_PRIMARY2 as foreground colours for shutdown screen (i.e. the same as the top bar background colour and top bar icon colour) unless using default theme. In that case, it uses a black background, and white foreground. 

Based on the results of the poll on Discord, of the ~35 respondents, 31 were in favour, and none not in favour of the shutdown screen being themed. I believe the remain four were basically wo [... cut]
closed 2021-11-29 by pfeerick