EdgeTX/edgetx #880: Font size companion appimage.

nostromo123, 2021-10-07:
Just  loaded the companion appimage 'EdgeTX "Dauntless" 2.5.0-RC3'.

Companion and simulator working! Thank you for including Linux!

But one remark. Font size is quite small. Please see pic attached.

Left companion appimage, right companion  build from main two weeks ago.

My system - Debian Bullseye.

br KH

[image]

raphaelcoeffic, 2021-10-08:
Hmm, very interesting. I did not pay attention to that! It could related to the Qt version we are using, and how it is configured. Let’s see if I can find out how that works.

raphaelcoeffic, 2021-10-08:
@nostromo123 can you please check this? https://unix.stackexchange.com/questions/202588/how-can-i-set-the-default-font-size-for-all-qt5-apps/217935#217935

Maybe this is something we can set by default inside the AppImage if that works.

pfeerick, 2021-10-08:
`QT_FONT_DPI=96` will probably do it (first screenshot), or `QT_FONT_DPI=128` (second screenshot). `QT_SCALE_FACTOR` will also enlarge, but it will scale everything, and won't end well. Basically it seems like the auto DPI scaling is doing it's job, and making things smaller... 😆 

[image]
[image]

raphaelcoeffic, 2021-10-08:
@pfeerick we should really avoid fixed DPI settings! Your screen should deliver it's DPI settings ;-)

raphaelcoeffic, 2021-10-08:
> Sorry, but where do you apply those settings?
> 
> Tried to add export QT_FONT_DPI=128 in ~/.profile.
> 
> Tried to add /etc/profile.d/qt-hidpi.sh with that command.
> 
> No change here ..... and yes, I rebooted.
> 
> Do I have to change over from XFCE to LXQT to see an effect?

You can start by testing directly in terminal. Just prepend the env variable to the command to have anything added to the environment.

raphaelcoeffic, 2021-10-08:
> An add. remark.
> 
> With companion, I build here from the source, using the qt5ct tools, I can chance the font size instantly, but as mentioned before, qt5ct does not show any influence with the appimage appearance. So for the appimage I'll follow pfeericks advice. Also helps with another appimage issue, I have here. On 'avidemux_2.7.8.appImage' the font size was also too small. This is fixed now!
> 
> br KH

Nice! This sounds like something we should document! Out of curiosity, can you please find out the real DPI resolution of your monitor?