EdgeTX/edgetx #885: 2.5 RC3 Erratic Mixer behaviour

Ga1j1n, 2021-10-09:
A model previously configured and functioning under 2.4 now exhibits un expected channel mixing after update, reverting restores normal function

The Issue was discovered on a EDF model with LS And RS slider outputs to Flaps and gear, after the update flaps and aileron channel output were mixed and gear was relocated to the flaps slider with no action from gear, all mixers appear sound otherwise
JUMPER T18 v1 (None touch) latest Multiprotocol firmware Channel Order RETA radio and module

Example model T45 Goshawk
[https://drive.google.com/file/d/16bHiMwmLPaWOK4knbP1PHNM33igeBCVx/view?usp=sharing](url)

raphaelcoeffic, 2021-10-09:
> A model previously configured and functioning under 2.4 now exhibits un expected channel mixing after update, reverting restores normal function
>
> The Issue was discovered on a EDF model with LS And [...]

Could you maybe just place the files in a ZIP file and upload here? Your link is asking me to request access... Apart from that, I'm not sure I understand quite the issue, but we'll see once the files are available.

Ga1j1n, 2021-10-10:
[Goshawk T18_2.zip](https://github.com/EdgeTX/edgetx/files/7317705/Goshawk.T18_2.zip)

Sure, here you go

Essentially, after flashing the radio to 2.5 rc3 the it appears the outputs that equate to flaps and ailerons have become joined (moving ailerons now also moves the flaps) the gear channel no longer operates no matter if I attempt to adjust the mixes, it as if the channel allocation has gone awry

closed 2021-11-08 by pfeerick