EdgeTX/edgetx #914: crashed RM TX16S with touchscreen in test lua script

MNS26, 2021-10-17:
**Describe the bug**
RM TX16S touchscreen breaks lua script and reopening it crashes the radio 

**To Reproduce**
https://drive.google.com/file/d/1tUkWwpORuFSGrMxJNBBed3axxf3IuZDo/view?usp=sharing

**Expected behavior**
for it to ignore touch event if not added to the script

**Screenshots**
(see video)

**General information**
 - EdgeTX version: 2.5.0 (28cdb40e)
 - Hardware: RM TX16S

MNS26, 2021-10-18:
@raphaelcoeffic i have a feeling we're looking at multiple different crashes as if they are ene.

I'm sure the touchscreen issue in standalone lua scripts  is definitely one of them.
But i feel like the USB storage mode might be an issue with files/scripts being loaded and then not being closed properly when entering USB storage mode.

MNS26, 2021-10-18:
@jfrickmann nope it still crashed (USB and touch)

MNS26, 2021-10-18:
i also found something else on the latest master branch (not related to this problem)
im assuming its supposed to be XXhXXm 
 
[image]

pfeerick, 2021-10-18:
Keeping in mind touch support hasn't been added to standalone scripts (only widgets),  I'm not sure how that can even be a factor here! :-O Regardless, something is not working, we're just not 100% clear on the what yet.

pfeerick, 2021-10-19:
Excellent. `[ViewMain] Focus lost` was what I was expecting to see. In the next day or so it will be a matter of running the code up again with the debugger attached and seeing why it is jumping out / losing focus...

MNS26, 2021-10-19:
i havent been able to track down the issue with USB yet, but im very suspicious of how it handles files that are open when going into storage mode

MNS26, 2021-10-19:
the USB craash is not related to just the ELRS script, ive crashed it with just a test one i made 

```
local function run(event)
  lcd.clear()
  lcd.drawText(0, 0, "Hello World")
  return 0
end

return { run = run }
```

MNS26, 2021-10-19:
@pfeerick pfff... finally got it to work but using the left uart  (to keep USB SD) but all i see is garbage on the serial port
[image]

rotorman, 2021-10-19:
This looks like wrong baudrate. Is it TX16S? Try 400.000 baud 8N1 on PC side.

pfeerick, 2021-10-19:
Whatever terminal or adapter you're using probably doesn't support that board rate, as 400000 baud is correct. Try putty (I use [kitty](https://github.com/cyd01/KiTTY/), a derivative of putty).

dlktdr, 2021-10-21:
Had a bit of grift getting mine to work right away too. I also have a CP2102. Is it a recent driver?https://www.silabs.com/documents/public/software/CP210x_Universal_Windows_Driver.zip

I use terra term and it shows more available serial bauds. You can set targets/horus/board.h LN# 670 to match one of the available bauds if 400,000 isn't listed. 

Can also check this link too http://www.olliw.eu/storm32bgc-wiki/How_to_configure_CP2102_USB_adapters_for_high_baud_rates the utility link isn't working but here is a copy https://www.silabs.com/documents/public/example-code/AN721SW.zip

MNS26, 2021-10-21:
@pfeerick the issue is that somehow the data becomes garbage so "just use putty" isnt that easy because it will only output _valid_ ascii characters but.... **if none of the data is valid.....** nothing will show

(furiously scrolled with the roller for 5 sec and.... nothing at all)
[image]

pfeerick, 2021-10-21:
Hm... that makes no sense... you logic analyser says the bits are the right way around, and that hterm is seeing them the wrong way around... but if putty is doing it also, it looks like your uart is somehow reading in wrong LSB/MSB order?

MNS26, 2021-10-21:
is there a way to... oh idk.. force a low baud rate (something like 115200)?
if i use serial over usb-c it works but i cant test the usb SD crash then

MNS26, 2021-10-21:
there we go...
[image]

i made the firmware use 115200 for debug

raphaelcoeffic, 2022-10-16:
This pretty old and has not been reported since then, so let's assume this has been fixed in a later release.

closed 2022-10-16 by raphaelcoeffic