EdgeTX/edgetx #915: No EVT_VIRTUAL_NEXT / EVT_VIRTUAL_PREV / ... for fullscreen widgets on TX16s

wimalopaan, 2021-10-17:
Title says all.

philmoz, 2024-08-20:
I think this can be closed. The key events are present in 2.10 and 2.11.

pfeerick, 2024-08-21:
Widget code to demonstrate it indeed is working (tested on TX16S, 2.10.3):
```lua
local name = "iss915"
local options = { 
}

local function create(zone, options)
 local widget = {
    zone = zone,
  }
  return widget
end

local function update(widget, options)
  widget.options = options
end

local function refresh(widget, event, touchState)
  if event == nil then -- Widget mode
    lcd.drawRectangle(0, 0, widget.zone.w, widget.zone.h)
    lcd.drawText(widget.zone.w / 2, widget.zone.h / 2, "Fullscreen!", DBLSIZE + CENTER + VCENTER + COLOR_THEME_PRIMARY3)
  else --fullscreen
    if event == EVT_VIRTUAL_EXIT then 
      print("Script exit executed")
      exit = 1
    elseif event == EVT_VIRTUAL_NEXT then
      lcd.drawText(40, 20, "EVT_VIRTUAL_NEXT", DBLSIZE + COLOR_THEME_PRIMARY3)
      print("Next!")
    elseif event == EVT_VIRTUAL_PREV then
      lcd.drawText(40, 20, "EVT_VIRTUAL_PREV", DBLSIZE + COLOR_THEME_PRIMARY3)
      print("Prev!")
    end
  end
end

return {
  name = name,
  options = options,
  create = create,
  refresh = refresh,
  update = update
}
```

closed 2024-08-21 by pfeerick