EdgeTX/edgetx #917: EdgeTX Companion - build radio firmware by selected options like OpenTX

IJK-CZ, 2021-10-18:
When this feature is scheduled? Currently, there is available a set of prebuilt firmwares (as *.zip file with set of firmwares), but for any other option (like different firmware language), individual firmware compilation is required. In OpenTX, it is part of Companion. In EdgeTX companion, the firmware options like 'Noheli' are available, but not download of firmware, passing those requirements.

JayPi01, 2021-11-16:
@pfeerick Tried to compile on Ubunto 20.04 but I had no success. Can you build me a firmware of the latest stable release with:-DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DBLUETOOTH=YES -DINTERNAL_GPS=NO

JayPi01, 2021-11-16:
@pfeerick I just found a version with my requirement so its not necessary to build one.

jpmarotta, 2021-12-30:
@pfeerick  Used the docker image to build custom firmware for my TX16S, but for whatever reason when I go to flash it via EdgeTX Companion, it tells me the the firmware is incompatible with the radio.  I've attached the options file I used, as well as the firmware. Any chance I could get you to take a look?

Specifically looking to turn on the following options:
- PPMUS
- NOHELI
- LUA
- EXTERNALACCESSMOD
- FLEX9M

[edgetx-v2.5-cmake-options.txt](https://github.com/EdgeTX/edgetx/files/7794492/edgetx-v2.5-cmake-options.txt)
[edgetx-v2.5_tx16s_lua-ppmus-mode2-R9M-ACCESS-noheli_release.bin.zip](https://github.com/EdgeTX/edgetx/files/7794497/edgetx-v2.5_tx16s_lua-ppmus-mode2-R9M-ACCESS-noheli_release.bin.zip)

IJK-CZ, 2022-01-25:
With version 2.6 still no change? For real usage of EdgeTX it should be supported feature.

gagarinlg, 2022-01-25:
Every person contributing is doing this as a hobby, not paid work. You can volunteer to set up and maintain a build server. Many people are using EdgeTX without problems, some even ask for a specific language version or other build options via discord or other means and they get it.