EdgeTX/edgetx #976: LUA: model.setFlightMode() behaves weird setting trims

wimalopaan, 2021-10-30:
I tried to exclude the trims 5 and 6 (index 4 and 5) of flightmode 0 and 1.
Exclude has the mode value 31. 
Doing so via model.setFlightMode(...) sets all (!) trims except index 0 (trim1) to exclude from that mode.

lshems, 2021-10-30:
Check on the opentx GitHub. Lots of info on this bug there

wimalopaan, 2021-10-30:
Well, if I read that correct, I have to pass the full table of trimsMode. Therefor I read the full table via getFlightMode(), change the two trims (index 4 and 5) to value 31, and then write it back. That should be ok, but isn't...

lshems, 2021-10-30:
The bug in opentx is that they made some signed integer error, AND don't use proper array indexing

wimalopaan, 2021-10-30:
Sounds like you have a fix for this?

lshems, 2021-10-30:
Would love to, bit can't program in c

wimalopaan, 2021-10-31:
It looks not to be possible to modify the table trimsModes returned getFlightMode().

As a workaroung one has to create a new table with six entries, and assign this table-ref to trimsModes and then write it back with setFlightMode().

lshems, 2022-10-15:
This is still not fixed, and should not be much work. Any chance ....

raphaelcoeffic, 2022-10-16:
@lshems @wimalopaan Gentlemen, please provide feedback on the PR fixing this issue (see #2567)

PR #2567 (2022-11-30): fix(lua): array indexes in some set/get functions
## Description

**Warning: this PR introduces a change in behaviour**

Prior to this change, `model.getFlightMode()` and `model.getCurve()` would return arrays with integer indexes starting at `0`, which breaks a lot of things with LUA. For instance, copying such an array removes the first element (index `0` is not copied).

**Summary of changes:**
- fixed array indexes in `model.getFlightMode()`
- fixed array indexes in `model.getCurve()`
- use array indexes to allow for incomplete table in `model.setFlightMode()`

Fixes #976

## Test LUA scri [... cut]
closed 2022-11-30 by pfeerick