EdgeTX/edgetx #977: Redirect inverted signal to Power up/down module bay on power modded Nirvana NV14

fpvdoc, 2021-10-30:
**Is your feature request related to a problem? Please describe.**
With a power modded Nirvana NV14 the Module Bay gets only 1.5 Volt on EdgeTX.

**Describe the solution you'd like**

In the original Firmware the stock Stepup Voltage Regulator get’s an inverted signal to enable/disable the module bay. That disables the stepup but doesn’t fully shut off the JR bay. It still gets battery voltage e.g 3.8. With the modded firmware the inverted signal is redirected to Testpin CH1.

The complete code is here:
github.com/open-flysky/Flysky-OpenTX-Test-Branch

Find the relevant changes for the power mod here - this code is ok for all variants with and without the HW mod:
\src\targets\nv14\exmodule_driver.cpp
Search for EXTMODULE_PWR_FIX_GPIO_PIN - this is set inverted to EXTMODULE_PWR_GPIO_PIN
The GPIO is defined in \src\targets\nv14\hal.h

Quellcode
void EXTERNAL_MODULE_ON()
{
  GPIO_SetBits(EXTMODULE_PWR_GPIO, EXTMODULE_PWR_GPIO_PIN);
  GPIO_ResetBits(EXTMODULE_PWR_FIX_GPIO, EXTMODULE_PWR_FIX_GPIO_PIN);
}
void EXTERNAL_MODULE_OFF()
{
  GPIO_ResetBits(EXTMODULE_PWR_GPIO, EXTMODULE_PWR_GPIO_PIN);
  GPIO_SetBits(EXTMODULE_PWR_FIX_GPIO, EXTMODULE_PWR_FIX_GPIO_PIN);
}
void extModuleInit()
{
  GPIO_InitTypeDef GPIO_InitStructure;
  GPIO_InitStructure.GPIO_Pin = EXTMODULE_PWR_GPIO_PIN;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_OUT;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_2MHz;
  GPIO_InitStructure.GPIO_OType = GPIO_OType_PP;
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP;
  GPIO_ResetBits(EXTMODULE_PWR_GPIO, EXTMODULE_PWR_GPIO_PIN);
  GPIO_Init(EXTMODULE_PWR_GPIO, &GPIO_InitStructure);
  //for additional transistor to ensuring module is completely disabled
  GPIO_InitStructure.GPIO_OType = GPIO_OType_OD;
  //pin must be pulled to V+ (voltage of board - VCC is not enough to fully close transistor)
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_NOPULL;
  GPIO_InitStructure.GPIO_Pin = EXTMODULE_PWR_FIX_GPIO_PIN;
  GPIO_SetBits(EXTMODULE_PWR_FIX_GPIO, EXTMODULE_PWR_FIX_GPIO_PIN);
  GPIO_Init(EXTMODULE_PWR_FIX_GPIO, &GPIO_InitStructure);
  GPIO_InitStructure.GPIO_Pin = EXTMODULE_TX_INVERT_GPIO_PIN;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_OUT;
  GPIO_Init(EXTMODULE_TX_INVERT_GPIO, &GPIO_InitStructure);
  GPIO_SetBits(EXTMODULE_TX_INVERT_GPIO, EXTMODULE_TX_INVERT_GPIO_PIN);
}

This is the power mod that completely shuts down the module bay and moreover allows the 18650's to drain down to 3.2 Volt.

https://docs.google.com/document/d/e/2PACX-1vRy9U54N_76afuX7yOfTy7nqqF62sL8To3QSVMR0jCer2gfv5SF3BVrcI_PaiOqJ6d9yqaJ7veFAzQn/pub

I'm pretty new to github so please forgive me if I miss something creating this issue.

stefan73, 2022-01-08:
Needs to be reopend. Not fixed (fyi: I am one of the guys how did the power mod documetn). Just tried edgetx 2.5 - does not work.
Details: CH1 pin (EXTMODULE_PWR_FIX_GPIO_PIN) not implemented as in the NV14 fork. Code is pasted above in this thread.

raphaelcoeffic, 2022-01-09:
> Just tried edgetx 2.5

@stefan73 This feature is not in 2.5, please upgrade to 2.6.0 RC2.

pfeerick, 2022-01-09:
Oops... should have looked at *when* this was resolved.

closed 2022-01-09 by pfeerick