EdgeTX/edgetx #978: LUA: getFieldInfo() does not return IDs for trim buttons

wimalopaan, 2021-10-31:
As the title says is it not possible to get the id of e.g. trim-button-5-up as getFieldInfo("5u").

eshifri, 2021-11-02:
{MIXSRC_TrimT5, "trim-t5", "Aux trim T5"},
    {MIXSRC_TrimT6, "trim-t6", "Aux trim T6"},

wimalopaan, 2021-11-02:
I'm not looking for the trim values!

I'm looking for the IDs of the trim buttons 5u, 5d, 6u, and 6d.

eshifri, 2021-11-02:
You are right! I stand corrected.
But... :-)
Actually, "5u" is not a button  - it is button position.
getValue("5u") returns something on the radios, where it does not exist: e.g. NV14.
I suspect it is interpreted as number 5.

If I read the code correctly trim buttons are not available as switches to lua at all. (They, probably, should.)

wimalopaan, 2021-11-03:
Well, I think I messed up the concepts ;-) 

More or less obviously one has to distinguisk betwenn mixer-sources and -id (MIXSRC_... ) and switch-id (SWSRC_...). To confuse more, these number (Ids )ranges overlap.

Now I added a LUA_EXPORT(...) for the trim buttons and also logical-switches to their SWSCR_... definitions. Then one gets theses switch-ids via getFieldInfo() also. These switch-id then can be used everywhere where swicthes are allowed, e.g. flightmodes or defintion of logical-switches using trim-buttons or other logical-switches.

I named the switch-id "sl1" to "sl64" for the logical-switch-ids (not to confuse now with "ls1" to "ls64" as their corresponding mixer-source-ods) and "t5u", "t5d", ...

Looks like this works no to seemlessly insert flightmodes or logical-switches (using switches) via LUA.

wimalopaan, 2021-11-03:
If someone can confirm this I would create a branch and PR

wimalopaan, 2021-11-03:
attached the new LUA exports list

[lua_exports_t16.txt](https://github.com/EdgeTX/edgetx/files/7467499/lua_exports_t16.txt)

wimalopaan, 2021-11-03:
Forgot the other trims:
[lua_fields_t16.txt](https://github.com/EdgeTX/edgetx/files/7467791/lua_fields_t16.txt)

eshifri, 2021-11-03:
Why do we need to distinguish ls1 - ls64 and sl1 - sl64?
Please notice that this requires changes not only for TX16, but for all radios.

wimalopaan, 2021-11-03:
"ls1" to "ls64" can act as mixer sources, they toggle between -1024 and 1024. so getFieldInfo() returns the mixer-source id. But that is not usable as switchid e.g. in model.setFlightMode().

I add "sl1" to "sl64" as names for switch-ids (that are not source-ids, if you query the value of a switch-id via getValue() you alswas get 0), these switch-is can be entered where switches (only) are expected e.g. flightModes.

Perhaps it would be better to rename getFieldInfo() to getSourceInfo() only returning mixer-source-ids (MIXSRC_... in dataconstants.h) and introduce a new function getSwitchInfo() only returning switch-ids (SWSRC_... in dataconstants.h).

eshifri, 2021-11-03:
I'm not  sure about changing the name for getFieldInfo() _ it will break all existing scripts and introduce major incompatibility with OTX, but adding a separate function for switches may be a good idea.

eshifri, 2021-11-03:
@pfeerick it is your call.

pfeerick, 2021-11-04:
**Changing** name - sorry, not happening anytime soon, since compatibility would be out the window.  If you wish to *add* a more descriptive name, i.e., similar #953, where the old name is still present (making the PR title a little misleading), with the old name then potentially being depreciated and subject to removal at a major version point, I don't see a problem. Especially if it is done in conjunction with adding a similarly named function that adds missing functionality. As Jim has suggested by making https://github.com/EdgeTX/edgetx/projects/5, the LUA API could do with some cleanup. 

I do question naming them sl1-sl64 when there are ls1-ls64, but I don't have a better suggestion as to naming. Anyway, if that makes sense, do a PR and we can fine tune it if needed.

jfrickmann, 2021-11-22:
We are talking about two different name spaces here. 
E.g. "sa" is a *source*, and `getValue("sa")` returns zero if switch A is in the middle, -1024 if it is down, and 1024 if it is up.
E.g. "SA↑" is  a *switch position*, and it is true if switch A is up. But we do not have a Lua function neither for finding an ID from a switch position string, nor to find a switch position string from an ID. And what this means, is that Lua functions such as e.g. `getLogicalSwitch()` are fairly useless, as we get the IDs for V1, V2 and V3, but we do not know what they mean, and we cannot generate a list of choices for changing them.
In strhelpers.cpp there is a function `getSwitchPositionName()` that finds the switch position string from the ID. Use it directly to get the string from an ID, and loop from 0 to `SWSRC_RADIO_ACTIVITY` to get an ID from a string (after reversing the sign for a "!" prefix).
So we could add the functions `getSwitchId(positionName)` and `getSwitchName(switchId)` to the Lua API.

jfrickmann, 2021-11-22:
P.S. Maybe we should also add Lua the constants `CHAR_UP` and `CHAR_DOWN` to make it easier for people to create strings like e.g. "SA↑" and "SA↓" with the EdgeTX characterset.

wimalopaan, 2021-11-22:
We need to have the possibility to query switch-ids. That are those ids that are accepted in e.g. model.setFlightMode() to activate that mode. The above suggested modification is not to get any value of the switch. For that we have getValue(), but getValue() uses source-ids, not switch-ids.

The above modification is to get the switch-ids of the trim-buttons and all of the LS, so that these elements can be used where switch-ids are requested (model.setFlightMode() oder mode.setLogicalSwitch().["and"]).

This is the general prblem in EdgeTx / OpenTx that there are several sorts of id, that are all numbers in LUA, but are not interchangeble.

jfrickmann, 2021-11-22:
Sources and switches are two different things, and they are not interchangeable. A source returns a value between -1024 and 1024, and a switch returns the values TRUE or FALSE. They cannot be used interchangeably, and therefore, a source-id and a switch-id cannot be interchanged either.
What may be confusing you, is that e.g. a three-position physical switch A may be read either as a source "sa" with the possible values -1024, 0, and 1024, or as six possible switches: SA↑, SA-, SA↓, !SA↑, !SA-, and !SA↓, each of which return the values TRUE or FALSE depending on the position of switch A.
Think of a mixer line. It has an input source, which is processed and possibly added or multiplied to generate an output. It also has a switch, which determines if the mixer line is active or not. These two are not interchangeable.
A more tricky example is a logical switch of the type "a > x" which takes a source as input (instead of a switch like most other logical switches). But this particular type of logical switch compares the source value between -1024 and 1024 to a constant value and returns TRUE or FALSE. So it is really a conversion from a source value to a switch value.
If you want to generate a source value from a switch value, then you could do it with an INPUT with two lines, where the first line is activated by the switch.

jfrickmann, 2021-11-22:
With regards to the naming of switches for finding switch-ids, we should use the names that are already in place; both because that would be the least confusing, and also because that would be the least work for me now, and more maintainable going forward.
So if you want the switch-id for LS1, you would use `getSwitchId("LS01")`. For physical switches it is a little more tricky, as e.g. "↑" is a Unicode character that is not available in the ASCII characterset used by Lua, and it is assigned a custom character number 202 in EdgeTX. So we have to either use `getSwitchId("SA\202")` or define character constants as in my "P.S." above, so we can use e.g. `getSwitchId("SA" .. CHAR_UP)`.

wimalopaan, 2021-11-22:
Thnak you for your explanation.

I think your solution to get the switch-IDs via getSwitchId("LS01") or getSwitchId("!LS01") and likewise for physical switches is the way we should go!

Thank you very much for your effort!.

jfrickmann, 2021-11-25:
@wimalopaan please check #1154!

closed 2021-12-01 by rotorman