EdgeTX/edgetx #981: Split Top Bar touch functionality into 2 boxes

audiogek, 2021-11-01:
For easier access to page two of the main menu.

Split the top bar touch box into two halves.
So when the right half is clicked/touched(red in picture) you go directly to page two.

[image]

jmxp69, 2021-11-06:
I have no influence on whether or not the developers will incorporate your request, but I thought I'd mention you can swipe from screen 1 to 2 by swiping. Just throwing that out there in case you didn't know.

audiogek, 2021-11-07:
I'm complete noob at github but as far as I can make out it looks like my request is already coded and tested.
I think I need to thank RCVideoReviews and close it or am I misreading this?

raphaelcoeffic, 2021-11-07:
> I'm complete noob at github but as far as I can make out it looks like my request is already coded and tested.
> I think I need to thank RCVideoReviews and close it or am I misreading this?

This will be closed auto-magically if the pull request gets merged.

pfeerick, 2021-11-07:
Coded by @eshifri, but not tested or merged yet. And as Raphael said, the issue will auto-close when the code if and when the PR is merged ;)

audiogek, 2021-11-07:
> Coded by @eshifri, but not tested or merged yet. And as Raphael said, the issue will auto-close when the code if and when the PR is merged ;)

Cool, thanks guys!

closed 2021-11-12 by pfeerick