EdgeTX/edgetx #994: XSR-SIM USB Dongle help

santacruzmofo, 2021-11-06:
I was having problem's being able to use velocidrone with the XSR-SIM USB Dongle and a radiomaster TX 16s. I got it to work on OpenTx by using the internal module and one of the Frsky settings. I can't rember which one... But if I was able to choose the internal module it would help! I'd at least be able to find the proper settings again. Unfortunately I can't find the internal module at all. Any help with this would be awesome! Is there anything I can put on my SDcard or LAU scripts I can run to get this accomplished?I LOVE EdgeTX and the ability to use the touchscreen, I also truly appreciate all of the developers hard Work! Please keep at it.

pfeerick, 2021-11-07:
I haven't used the XSR-SIM myself, but the specs state "Compatible with FrSky transmitters and module systems (Mode D16)" ... so you are most likely want to set the Internal RF (near the bottom of model settings) to MPM, with Frsky as the protocol and D16 as the subprotocol. Then bind it and it should just work.

santacruzmofo, 2021-11-07:
Thank you for your quick response, I will try this today and let everyone know how it goes!

rburrow87, 2021-11-08:
I use mine with ACCST2, so you would need to choose FrSkyX2 and D16 in the protocol list. I can't remember if it came that way or if I flashed it to ACCST2 though.