EdgeTX/edgetx #1004: Empty Screen Flickering

MuhammadQaisarAli, 2021-11-07:
**Describe the bug**
If a screen is added, the single widget is selected and all the check boxes are unchecked, which means a transparent screen, then the screen starts to flicker and makes the system unstable.

**Screenshots**
https://user-images.githubusercontent.com/45729347/140655833-f7acff68-748d-466a-995b-6ca8f3670813.mp4

**General information**
 - EdgeTX version 2.5
 - Hardware RM TX16S

MervDale, 2021-11-10:
Yes it is repeatable in that if I follow the along the same lines I can get the menu to get into a constant flickering mode, all though not always, sometimes just 3 or 4 flicker before  it settles down

raphaelcoeffic, 2021-11-10:
That looks like the back buffer is not wiped prior to display, thus showing the old image.

MervDale, 2021-11-11:
I must say the latest nightly seems to fix some of the issue with the flickering and  that a screen swipe doesn't seem to work when you have double tapped a widget and it also seems to have stopped the flickering on the GPS lua script that I was watching

pfeerick, 2021-11-11:
Oh, fancy... anyone else get those disco vibes? 😆 I wonder if this is also related to the screen flicker you get when editing widgets.. seemes to have a bit of a panic attack there also with the right sequence of actions...

gagarinlg, 2022-07-20:
Does this still happen? My personal opinion is to close this and make a new issue, when this still happens with the new UI

pfeerick, 2022-07-21:
Funny you say that... I came across this issue again yesterday or the day before, and was of the same mind, as there was a lot of changes around screen buffering so this should be gone, but I'd not been using the nightlies/LVGL stuff enough of late to be able to concretely say it was gone. So, closing until it resurfaces since LVGL and associated changes appear to resolved it.

closed 2022-07-21 by pfeerick