EdgeTX/edgetx #1008: NV14: R9 FLEX unavailable in menu even when firmware built with MODULE_PROTOCOL_FLEX=ON

shellixyz, 2021-11-07:
**Screenshots**

[image]

**General information**
 - EdgeTX release 2.5.0 and main as of today built with  MODULE_PROTOCOL_FLEX=ON
 - Hardware FlySky Nirvana / NV14

pfeerick, 2021-11-08:
Have you tried both options anyway to see if it is cosmetic only? i.e. is FCC 868Mhz and EU 915Mhz ? 

I see also that the S.Port link state/option message isn't being displayed either... i.e. as shown here... https://youtu.be/OgvgzPm6moE?t=210

shellixyz, 2021-11-08:
@pfeerick It is definitely not only cosmetic. Flex should have options to push the power to 1W with telemetry but it is not possible. In any case it is not because Flex is enabled that it should render FCC and LBT modes unusable.

Today I also flash my T16 and the same problem here so it is not NV14 specific like I originally thought. I guess all radios or perhaps only Horus type radio are impacted

[image]

[image]

[image]

raphaelcoeffic, 2021-11-08:
@shellixyz i was thinking about removing that option and just enabling it by default, what do you think?

shellixyz, 2021-11-08:
> @shellixyz i was thinking about removing that option and just enabling it by default, what do you think?

Sounds good to me. I wonder why it was made an option back when it was introduced

raphaelcoeffic, 2021-11-08:
> > @shellixyz i was thinking about removing that option and just enabling it by default, what do you think?
> 
> Sounds good to me. I wonder why it was made an option back when it was introduced

Probably for radios shipped by FrSky to be compliant with some regulations. However, it all depends on the firmware installed on the module, so why bother?

shellixyz, 2021-11-08:
> Probably for radios shipped by FrSky to be compliant with some regulations. However, it all depends on the firmware installed on the module, so why bother?

Indeed must be the reason and since they don't ship their modules / radios with FLEX firmware it doesn't matter

PR #1043 (2021-11-28): Remove FLEX options (and make it default)
Fixes #1008
closed 2021-11-28 by pfeerick