EdgeTX/edgetx #1014: NV14 - Fullscreen lua widgets and input

pfeerick, 2021-11-08:
Fullscreen (lua) widgets don't get the emulated trim hat input benefits that #985 brings, so need some consideration as to how they will be interacted with on the NV14 (and also future potential targets like the PL18, since that has normally position trims, so isn't as... practical to emulate inputs via trims). It *is* possible to have it so that the widgets toggle the trims/key state as part of this function, but therein lies the problem of what if you want to use trims when a widget is fullscreen. Ideally, the real solution moving forward is getting all the lua widgets to adopt touch support since it is available to them. So this is mostly a discussion and reminder issue. 

https://github.com/EdgeTX/edgetx/blob/e6697639c7c3de3c770d70056ead096d60e53a46/radio/src/gui/colorlcd/widget.cpp#L162-L187

eshifri, 2021-11-08:
Can we use pushEvent() for HW key, when touch is detected?

pfeerick, 2023-12-07:
Lua scripts to support touch events to some extent now, and with the trims as keys emulation #3894, it should also be possible to navigate scripts using trims, so this issue should no longer be relevant.

closed 2023-12-07 by pfeerick