EdgeTX/edgetx #1015: On screen keyboard "remembers" state

NemesisXB, 2021-11-08:
**Describe the bug**
The keyboard remembers the last state it was in. If you edit a field and for example switch to the numeric input and finalise the input. If you then select another (or the same) field to edit, the keyboard will still be in the numeric input mode.

**Expected behavior**
The keyboard should start up in the same state every time.

**General information**
 - EdgeTX version 2.6.0 809db515
 - RM TX16S

pfeerick, 2021-11-08:
Another minor 'annoyance' to me is the caps key... It would prefer it like a phone shift key, single touch shift (i.e. released on letter press), second touch lock (stays locked till released), third touch release.

NemesisXB, 2021-11-09:
> Another minor 'annoyance' to me is the caps key... It would prefer it like a phone shift key, single touch shift (i.e. released on letter press), second touch lock (stays locked till released), third touch release. I think it stays set between fields also.

Yes it also stays set between fields.

pfeerick, 2021-11-09:
Nah, ask away, sometimes it just hasn't been noticed / thought of / ignored for now ;)

gagarinlg, 2022-07-20:
Keyboard has been replaced with LVGL keyboard

closed 2022-07-20 by gagarinlg