EdgeTX/edgetx #1016: Trainer inputs extended to 8 channels for FPV headtracking transmission

johaninpika, 2021-11-08:
i m using googles with headtracking supported over usb output. i make RaspberiPI based translator to ch5 and ch6   to JR connector. Now I realize only 1 to 4 channels available. Can someone add option to read more channels on trainer lnput, please. Using Nirvana radio and finaly edgetx has this great promise to move further some projects...

any other option, maybe over usb input??
best regards and thanks 

*Before suggesting a new feature here, please search the open issue if there is already a request for the feature.
You can add your comment there, or react so that we know know much interest there is in the feature request.*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

raphaelcoeffic, 2021-11-08:
No, you can use TRx inputs into sources and mixers. The first 4 channels are a bit more convenient to use, as it covers the most “normal” use case.

gagarinlg, 2023-02-26:
closed, because it is possible to use more than 4 channels

closed 2023-02-26 by gagarinlg