EdgeTX/edgetx #1024: x9d+ gimbals stop working

bigrip, 2021-11-10:
hi first i flashed the wrong bootlader x9d+ 2019 instead of x9d+,
then i flashed the x9d+ bootlader and installed edgetx but my gimbals are not longer working.
then i flashed open tx but my gimbals are no longer working.
pls help me to get my radio working again

raphaelcoeffic, 2021-11-11:
Have you tried to re-calibrate the gimbals?

bigrip, 2021-11-11:
the radio is not responding to gimbal input or button´s , trimm buttons work.

raphaelcoeffic, 2021-11-11:
> the radio is not responding to gimbal input or button´s , trimm buttons work.

You have most probably still the wrong target installed. If the menu / page / + / - don't work, this is the most probably cause. Please join us at our Discord server so we can sort it out with you.

closed 2021-11-11 by raphaelcoeffic