EdgeTX/edgetx #1027: Horus X12s - No Flash Boot Loader Option

Portfox, 2021-11-11:
Have run into a bit of a snag, copied the .bin file across to the FIRMWARE folder and pressed SYS to select Flash Bootloader, but I only get Copy Paste or Delete as an option.
Any ideas what i have done wrong?

Steve

The issue tracker here is for devlopment of EdgeTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/), [Discord](https://discord.gg/wF9wUKnZ6H)
or [Facebook](https://www.facebook.com/groups/edgetx).

Issues that fall into the category of asking for help may be re-directed
to the other support platforms or answered directly if time permits.

pfeerick, 2021-11-12:
@Portfox Can you give this a try - you'll just have make sure you pick the right firmware for your transmitter ;) => https://github.com/EdgeTX/edgetx/suites/4325378447/artifacts/113210536

Portfox, 2021-11-12:
Thanks Peter
Its very strange, when the X12s is plugged in via USB, the drives do not show on the PC. It did show originally.
I can write the Edge software to the SD card via a SC card reader and then plug it into the X12s, then boot the X12s connect the USB cable and try to flash using the Flasher, it just give me errors. (i've attached part of the flasher log below)
Also gave that other .bin loader a go as well, but if its added to the [FIRMWARE] folder, there is no option to load (flash) it.

Ignore drives D, H and Z, they are my network drives, and i'm on a domain. That shouldnt matter.

[created.message][2021-11-12T01:13:07.454Z] Instance created, loading info from GH API
[updateContent.data.indexdat][2021-11-12T01:13:12.152Z] {"targets":[["Flysky NV14","nv14-"],["Frsky Horus X10","x10-"],["Frsky Horus X10 Access","x10-access"],["Frsky Horus X12s","x12s-"],["Frsky QX7","x7-"],["Frsky QX7 Access","x7-access"],["Frsky X9D","x9d-"],["Frsky X9D Plus","x9dp-"],["Frsky X9D Plus 2019","x9dp2019-"],["Frsky X9 Lite","x9lite-"],["Frsky X9 Lite S","x9lites-"],["Frsky X-Lite","xlite-"],["Frsky X-Lite S","xlites-"],["Jumper T8","t8-"],["Jumper T12","t12-"],["Jumper T16","t16-"],["Jumper T18","t18-"],["Jumper T-Lite","tlite-"],["Radiomaster TX12","tx12-"],["Radiomaster TX16s","tx16s-"]],"changelog":"- Major initial bug fixes"}
[flashFw.message][2021-11-12T01:13:17.420Z] Firmware file cache deleted
[flashFw.message][2021-11-12T01:13:17.424Z] Start DFU Util binary...
[flashFw.message][2021-11-12T01:13:17.424Z] -a 0 --dfuse-address 0x08000000 --device 0483:df11 -D C:\Users\portfox.PORTFOXDEVELOPM\AppData\Roaming\edgetx-flasher\flash.bin
[dfuUtil.message][2021-11-12T01:13:17.426Z] Found DFU Util for Windows, path set.
[created.message][2021-11-12T01:14:37.558Z] Instance created, loading info from GH API
[updateContent.data.indexdat][2021-11-12T01:14:49.967Z] {"targets":[["Flysky NV14","nv14-"],["Frsky Horus X10","x10-"],["Frsky Horus X10 Access","x10-access"],["Frsky Horus X12s","x12s-"],["Frsky QX7","x7-"],["Frsky QX7 Access","x7-access"],["Frsky X9D","x9d-"],["Frsky X9D Plus","x9dp-"],["Frsky X9D Plus 2019","x9dp2019-"],["Frsky X9 Lite","x9lite-"],["Frsky X9 Lite S","x9lites-"],["Frsky X-Lite","xlite-"],["Frsky X-Lite S","xlites-"],["Jumper T8","t8-"],["Jumper T12","t12-"],["Jumper T16","t16-"],["Jumper T18","t18-"],["Jumper T-Lite","tlite-"],["Radiomaster TX12","tx12-"],["Radiomaster TX16s","tx16s-"]],"changelog":"- Major initial bug fixes"}
[flashFw.message][2021-11-12T01:14:57.923Z] Error in deleting firmware file
[flashFw.message][2021-11-12T01:14:58.555Z] Start DFU Util binary...
[flashFw.message][2021-11-12T01:14:58.555Z] -a 0 --dfuse-address 0x08000000 --device 0483:df11 -D C:\Users\portfox.PORTFOXDEVELOPM\AppData\Roaming\edgetx-flasher\flash.bin
[dfuUtil.message][2021-11-12T01:14:58.556Z] Found DFU Util for Windows, path set.
[flashFw.message][2021-11-12T01:15:23.745Z] Firmware file cache deleted
[flashFw.message][2021-11-12T01:15:23.750Z] Start DFU Util binary...
[flashFw.message][2021-11-12T01:15:23.750Z] -a 0 --dfuse-address 0x08000000:unprotect:force --device 0483:df11 -D C:\Users\portfox.PORTFOXDEVELOPM\AppData\Roaming\edgetx-flasher\flash.bin
[dfuUtil.message][2021-11-12T01:15:23.752Z] Found DFU Util for Windows, path set.
[flashFw.message][2021-11-12T01:16:23.201Z] Error in deleting firmware file
[flashFw.message][2021-11-12T01:16:23.215Z] Start DFU Util binary...
[flashFw.message][2021-11-12T01:16:23.215Z] -a 0 --dfuse-address 0x08000000:unprotect:force --device 0483:df11 -D C:\Users\portfox.PORTFOXDEVELOPM\AppData\Roaming\edgetx-flasher\flash.bin
[dfuUtil.message][2021-11-12T01:16:23.217Z] Found DFU Util for Windows, path set.
[flashFw.message][2021-11-12T01:16:42.394Z] Firmware file cache deleted
[flashFw.message][2021-11-12T01:16:42.396Z] Start DFU Util binary...
[dfuUtil.message][2021-11-12T01:16:42.397Z] Found DFU Util for Windows, path set.
[created.message][2021-11-12T01:17:15.509Z] Instance created, loading info from GH API
[updateContent.data.indexdat][2021-11-12T01:17:22.692Z] {"targets":[["TX16s","tx16s-"],["QX7","x7-"],["X9D","x9d-"],["X9D Plus","x9dp-"],["X9D Plus 2019","x9dp2019-"],["T-Lite","tlite-"],["X-Lite","xlite-"],["X-Lite S","xlites-"],["X9 Lite","x9lite-"],["X9 Lite S","x9lites-"],["T16","t16-"],["T18","t18-"],["TX12","tx12-"],["Horus X12s","x12s-"],["Horus X10","x10-"],["T12","t12-"],["T8","t8-"]],"changelog":"- Major initial bug fixes"}
[flashFw.message][2021-11-12T01:17:28.510Z] Error in deleting firmware file
[flashFw.message][2021-11-12T01:17:28.533Z] Start DFU Util binary...
[flashFw.message][2021-11-12T01:17:28.534Z] -a 0 --dfuse-address 0x08000000 --device 0483:df11 -D C:\Users\portfox.PORTFOXDEVELOPM\AppData\Roaming\edgetx-flasher\flash.bin
[dfuUtil.message][2021-11-12T01:17:28.536Z] Found DFU Util for Windows, path set.
[updateConfig.message][2021-11-12T01:18:23.763Z] Scanned all block devices
[updateConfig.data][2021-11-12T01:18:23.764Z] [{"name":"C:","identifier":"C:","type":"disk","fsType":"ntfs","mount":"C:","size":"111761793024","physical":"Local","uuid":"B4F96F26","label":"OS","model":"","serial":"B4F96F26","removable":false,"protocol":""},{"name":"D:","identifier":"D:","type":"disk","fsType":"ntfs","mount":"D:","size":"1000068870144","physical":"Local","uuid":"92E2E9BD","label":"DATA","model":"","serial":"92E2E9BD","removable":false,"protocol":""},{"name":"H:","identifier":"H:","type":"disk","fsType":"ntfs","mount":"H:","size":"997487276032","physical": [... cut ...] ":"disk","fsType":"ntfs","mount":"H:","size":"997487276032","physical":"Network","uuid":"E0955F8E","label":"Shares","model":"","serial":"E0955F8E","removable":false,"protocol":""},{"name":"Z:","identifier":"Z:","type":"disk","fsType":"ntfs","mount":"Z:","size":"2948396015616","physical":"Network","uuid":"6646C5A0","label":"Public","model":"","serial":"6646C5A0","removable":false,"protocol":""}]
[created.message][2021-11-12T01:18:33.753Z] Instance created, loading info from GH API
[image]
[image]

pfeerick, 2021-11-12:
> boot the X12s connect the USB cable and try to flash using the Flasher

For DFU flashing (i.e. default for Flasher), unless the X12s is different to most other radios, you connect the USB with the radio off, and leave it turned off... as DFU is a different mode, and won't show any drive letters. 

Are you able to access the bootloader itself - i.e. trims held inwards while powering on the radio? If so, try flashing the firmware that way, and then see if the 'flash bootloader' entry is now present when running the firmware proper. 

Sorry if you've actually already done this... it's always hard to tell exactly a persons level of knowledge/what they've done ;)

Portfox, 2021-11-12:
Peter
Bingo!
The TX has to be off, I feel like a goose, didn't even think to do that. Thanks again for your help.

Now onto trying to fix why it keeps telling me the Battery is low, but its full.
Any thoughts on that one?

Thanks again, Steve

pfeerick, 2021-11-12:
> The TX has to be off, I feel like a goose, didn't even think to do that. Thanks again for your help.

Don't worry... I've done worse 😁 And no problem. So I take it now that you're running that firmware, that an option to flash the bootloader via the sd browser appears now?

So does the transmitter actually think the battery is low (i.e. on the radio/sys menu -> hardware page, there should be an entry for battery calibration that shows the voltage the transmitter thinks the battery is) - i.e. is it reading the voltage correctly? Perhaps it need recalibrating? Otherwise, no idea I'm afraid.

Portfox, 2021-11-12:
The battery is ok, I upgraded a while back to a Lipo system for it, thats fixed, just had to adjust the ranges as it differs from the Nicad standard battery.
I had the turn off the 'Çheck RTC Voltage' as it was complaining on boot up. But that could be just the coin battery needs replacing. Although that didn't do that before the update. I believe that just holds the time/date?

Steve

pfeerick, 2021-11-12:
Yes, RTC battery pretty much holds the time/date setting. If it didn't need it before the update (OpenTX?) and does now, it is could a bug with how the voltage is being read... that has happened for one or two other models ;)

pfeerick, 2021-11-13:
@Portfox Are you able to confirm that the "Flash bootloader" option now appears when selecting a firmware file in the sd card browser on the radio?

Portfox, 2021-11-14:
Sorry for the delay, yes, all good. It shows the correct firmware when holding the two trims and turning on the X12s.
Thanks again for all your help.

Just have the RTC alert when powering on the X12s, I can switch it off in the settings, but I'll leave it on to make me check for new updates in case there is a fix for it.
I'll copy my models back over and give it a run in a couple of weeks, Bit on at the moment,

pfeerick, 2021-11-15:
@Portfox No worries. Sorry, that's a different issue... What I'm after is when you go into the SYS/radio menu, and browse to the firmware folder in the SD card browser, do you get a "Flash bootloader" option when long pressing on a firmware bin file? ;)

pfeerick, 2021-11-15:
@brainbubblersbest you said this worked fine for you in #385 (and are actually responsible for it being fixed) ... is it still working for you now?

brainbubblersbest, 2021-11-15:
Need to try a latest build.
Im very busy the last weeks, and I stopped trying newest builds bc im
waiting still for access Menu options. Cant use my Radio without them.
But I will try.

PR #1033 (2021-11-20): std::string::data() gives no guaranty on 0-termination
Fixes #1027 

Summary of changes:
- use `.c_str()` instead of `.data()` when a 0-terminated string is used.
closed 2021-11-20 by pfeerick