EdgeTX/edgetx #1029: Incompleteness of Input Selection with Filtering

richardclli, 2021-11-11:
**Describe the bug**
Channel/Switch selection is incomplete when filter is used

**To Reproduce**
1. Choose a switch in Inputs, or Logical Switch screens, etc.
2. By default a complete list is shown
3. After tapping the filter tab, some choices in the default list is not shown, e.g. "---", "6P6", ... "6P1"

**Expected behavior**
Need to review the filtering and control.  Need to have a complete list e.g. "---" must be included, and or the default list maybe wrong.

**Screenshots**
[image]
[image]
[image]

**General information**
 - EdgeTX version 2.5.0
 - Hardware TX16S

pfeerick, 2021-11-11:
This issue elaborates on this somewhat also: https://github.com/EdgeTX/edgetx/issues/228

JimB40, 2021-11-11:
@pfeerick are you working now on popup list? #212

pfeerick, 2021-11-11:
This is the filters, not the menu ;)

JimB40, 2021-11-12:
Okay let's be on the same page with naming
What I see above is called in UI **list**
This particular list has additional filter options displayed as **buttons** 
Above shown list is part of **pop-up window**.
This particular pop-up window has no **window header** displayed.

Can you explain now what is the difference from code perspective? And what you call menu?
Someone did above as separate piece of code?

pfeerick, 2021-11-12:
Changing the appearance of the menu (which is what you are thinking of as to re : what I'm working on for the menu) is not the same as changing the behaviour of, or changing the function that generate the data to be displayed.

JimB40, 2021-11-12:
What you call "Menu"? :)

pfeerick, 2021-11-12:
They are all menus... (a popup window showing a list). If you set a title, it has a title (window header), if you set a toolbar, it has a toolbar on the side (buttons), which is being used as a filter. Now do you see why I simply circled the screen element in that issue and said "this isn't working" rather than worry about "what is it called"? 😆 

Changing the appearance generally means changes to libopenui. Changing what is displayed, and determined what data to display is something that needs to be changed in the relevant place. i.e. inputs source list gets changed on inputs page code, logical switches switch gets changed on logical switches page code, etc, etc. As I said, changing appearance is not the same as changing functionality or data that goes into it, so this is not the same or related issue ;)

JimB40, 2021-11-12:
I know that. just asking if we are talking about same piece of code.
So as I understand any changes to pop-up UI (colors etc) affect all pop-ups no mater if they display commands for field, available source list, available switches list etc

pfeerick, 2022-08-03:
This issue is still present in a0770beb5f8616e2f14f2828de12b3ff8b1f3397. It seems like it is also worse, in that it seems presses of `PAGE >` and `PAGE <` to switch between the filters either has a *lot* of lag, or is being ignored. Also, multiple filters are hilighted when pressed, instead of being mutually exclusive.

pfeerick, 2022-09-22:
And while we're at it... here is another filters related issue #2344