EdgeTX/edgetx #1030: Hardware analog input as switches

JyeSmith, 2021-11-11:
Is it possible to have the Hardware option to make sliders/pots any of the available switch options e.g. toggle, 2, or 3 position?

Sliders and pot options could also share the same Hardware options since they are just analog inputs e.g. either can be slider, pot, multiswitch.

Making the hardware more generic will allow users to mod handsets and abuse various inputs for other means... Like me who put a switch on the slider input because I had a plug that fit 😄 

Cheers

raphaelcoeffic, 2023-09-05:
@JyeSmith hehe, looks like you're finally getting that feature ;-)

JyeSmith, 2023-09-05:
Woohoo!  Thanks mate.

That'll make it easier throwing away useless sliders and replacing them with switches 😄

closed 2023-11-02 by pfeerick