EdgeTX/edgetx #1031: Jumper T-Lite contract bug

NiklasVoigt, 2021-11-11:
**Describe the bug**
OpenTX and EdgeTX both have the same Bug. In the Bootloader and when starting the Radio the contrast of the display is very low.
After showing the bootlogo the contrast is good. In Bootloader the contrast is always very low.

**To Reproduce**
Go to Bootloader.
Start the Radio.

**Expected behavior**
Normal contrast as after bootlogo.

**Screenshots**
[image]
[image]
[image]

**General information**
 - EdgeTX version 2.5.0 or OpenTX 2.3.14
 - Hardware Jumper T-Lite

pfeerick, 2021-11-12:
I'm glad it's not me... I've always thought the T-Lite display (and also the TX12) was not the best at times. i.e. the boot screen on the TX12 gets a bit darker during the splash delay, and the bootloader screen on the T-Lite has always been dim - it's as if it doesn't go fully black when it should.

erstec, 2021-11-15:
@NiklasVoigt can you test it on real T-Lite if I provide you binary?

erstec, 2021-11-15:
@NiklasVoigt
[edgetx-2.5.0-bl_contrast.bin.zip](https://github.com/EdgeTX/edgetx/files/7541020/edgetx-2.5.0-bl_contrast.bin.zip)

pfeerick, 2021-11-16:
This seems noticeably darker on my T-Lite. Is there something also going on just after the four dots / segments animation... as looking at it a bit more closely now, it seems like the dots are probably nomal darkness, and then just as the last one lights up, before the logo shows, the contrast seems to drop for the logo... and be normal again after that. Just thought I'd mention it incase it was a similar problem/fix needed.

erstec, 2021-11-16:
@pfeerick Ok, I see. Please try again, if you can.
[edgetx-2.5.0-bl_contrast1.bin.zip](https://github.com/EdgeTX/edgetx/files/7546145/edgetx-2.5.0-bl_contrast1.bin.zip)

pfeerick, 2021-11-17:
So, this does appear to work, but causes another problem... the T-Lite will not boot :-P 

You may be then scratching your head and asking how this fixes the problem if I can't turn the radio on... well... it won't boot if pressing the power button... backlight stays off, blue LED, nothing on the screen. But if I instead enter the bootloader, and exit via that, the contrast seems ok... 

Is it likely the change you made caused this, or should I ping Raphael to check if there are linker issues affecting the T-Lite target also (we just had one that b0rked the NV14 builds - was just a ticking time bomb waiting to go off).

NiklasVoigt, 2021-11-17:
If it won't boot now I won't flash the test files haha. The t lite is my daily driver..

erstec, 2021-11-17:
> If it won't boot now I won't flash the test files haha. The t lite is my daily driver.. I think @pfeerick has become our experiment animal haha

You can: start to bootloader, press Exit and it will boot

And please don't use last provided binary, I found issue in it and will provide fixed version shortly

erstec, 2021-11-17:
> So, this does appear to work, but causes another problem... the T-Lite will not boot :-P
>
> You may be then scratching your head and asking how this fixes the problem if I can't turn the radio on... [...]

I found what is wrong, it my fault, wrong compiler definitions... really sorry... fixed already and "last" try to solve mentioned problem FW placed below. Test it if you can. Thank you for support!

If it will not solve problem - at least we will have good contrast in BootLoader :)

erstec, 2021-11-17:
> If it won't boot now I won't flash the test files haha. The t lite is my daily driver.. I think @pfeerick has become our experiment animal haha

And I am really soory for this case. If you can;t solve this as I described before - please contact me on Discord ErNis#3947 and I will help you immediately!

erstec, 2021-11-17:
@NiklasVoigt @pfeerick Please use this binary. Testes on both my remotes, no 'freeze' issue anymore
[edgetx-2.5.0-bl_contrast3.bin.zip](https://github.com/EdgeTX/edgetx/files/7553102/edgetx-2.5.0-bl_contrast3.bin.zip)

pfeerick, 2021-11-18:
> really sorry... fixed already

Don't be... that's what testers are for ;)

Anyway, I think you are on to a winner with `contrast3`... bootloader looks fine, main firmware seems fine... no 'noticable' "lightening" of the dots or splash screen. Oh, and it does boot normally! 😁

PR #1099 (2021-11-21): Taranis-based B/W: Initial Contrast fix
Fixes #1031 

Summary of changes:
Taranis based radios with B/W screens now have LCD_CONTRAST_DEFAULT initialization in bootloader and normal startup.
This prevents 'unreadable (too low or too high contrast' screen in bootloader and 'blink' after 'dots' on Jumper T-Lite and other similar radios.
closed 2021-11-21 by raphaelcoeffic